EdgeTX/edgetx #1971: NV14: Negative value of altitude if the flight is above 300 meters

Ktr128, 2022-05-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The altitude value becomes negative if you take off above approximately 300 meters

### Expected Behavior

When taking off above 300 meters, the altitude should not become negative.

### Steps To Reproduce

Fly above 300 meters with the fs-cat01 altimeter

### Version

Other (Please specify below)

### Transmitter

FlySky NV14

### Anything else?

Fw.ver 2.7.1 
Altimeter : fs-cat01

Ktr128, 2022-05-16:
I emulated this bug with a vacuum cleaner:
https://youtube.com/shorts/wCUPsf5UMB4?feature=share

eshifri, 2022-05-16:
I do not see anything obvious in the code.
Can you please clarify:
1. 300 m - is it a ground height or the sign changes when you go above 300 m?
2. Do you happen to know what the pressure sensor was saying?
3. What was the temperature (approximately)?
4. Any chance the temperature (Centigrade) changed the sign with height?
5. Was it just the incorrect sign, or the value was complete garbage?
6. Did we have the same problem in NV14 OTX?

Ktr128, 2022-05-17:
1. The sign changes when the value passes a little more than 300 meters.
Earth altitude = 0 meters;

2. The pressure sensor shows the correct value, video:
https://www.youtube.com/watch?v=HgxIYUSzpnM

3. Temperature approximately 20 degrees Celsius;

4. There is no such probability;

5. As far as statistics allow, just a wrong sign;

6. NV14 OTX didn't have this problem.

eshifri, 2022-05-17:
Thank you!
Was it a vacuum cleaner that allowed to reach such altitude? :-)
Looks like there is a problem with temperature decoding.
There is a strange line in the end of the altitude calculation: if both temperature and log(pressure) are negative, change the altitude sign. The same code is present in OTX though...

Ktr128, 2022-05-17:
Yes, in the experiments I've been doing now, it was a vacuum cleaner. Without it, in aircraft modeling is impossible :)
But the first time I encountered this in a real flight, when I climbed above 300 meters.

Ktr128, 2022-05-17:
I don't understand what this line is for:
int32_t ln = t >> 31;
(file flysky_ibus.cpp, line 398)

Ktr128, 2022-05-17:
6. NV14 OTX didn't have this problem. - This is not the truth.

I remembered that for OTX I wrote a height detection script and used it:
local function init_func()
local mswitch_value = 0
end
local function run_func(event)
local myxx = getValue("Airp") -- getting pressure
myxx = myxx * 0.75
myxx = myxx * 12 --Altitude calculation without temperature
local zerox = model.getGlobalVariable(1, 0)
zerox = gzeroh * 0.75
zerox = zerox * 12
local hplane = (zerox - myxx)
local zvuk = model.getGlobalVariable(0, 0)
zvuk = zvuk + 1
if hplane > zvuk then
playFile("/SOUNDS/en/myup.wav")
end
zvuk = zvuk - 2
if hplane < zvuk then
playFile("/SOUNDS/en/mydown.wav")
end
model.setGlobalVariable(0, 0, hplane)
-- playNumber(hplane,5)
mswitch_value = 105
return 2
end
return { run=run_func, init=init_func }

Thus, I cannot say that the regular calculation worked in OTX.

eshifri, 2022-05-17:
I'm not sure. I suspect it is just a rescaling of the log() to normal range, but I have never checked the math there.
There are definitely problems with temperature in this code.
The temperature value, used in this function, are neither in Kelvin nor in Celsius.
If I believe the code in this file, line 183 says that  ((value >> 19) - 400), UNIT_CELSIUS gives us temperature in Celsius, so Kelvi, required for altitude calculation should be  ((value >> 19) - 673). getAlt() uses just  (value >> 19).
Thar being said, I have tried to add temperature as a sensor and had to comment it out, since there was no way to calibrate the sensor. I'm even not sure that valu>>19 is related to the temperature. Maybe we are using wrong bits.
I'm afraid the immediate solution is just to remove temperature from if (neg ^ tempNegative) result = result * -1;

Ktr128, 2022-05-17:
344 line:
 return (uint16_t) (tempertureIbus - 400) + 2731;

Looks like it's being converted to Celsius. I'm not sure if this is true

eshifri, 2022-05-17:
I missed it. Makes perfect sense now.
So the only question is whether value>>19 decodes relevant bits.

Ktr128, 2022-05-17:
helper variable is declared as uint_64t
after:
 int result = (int) helper;
And rusult is returned as int32_t
Is everything ok here?

Ktr128, 2022-05-17:
I did it!
As I felt at first that cpp knows how to joke when converting types.
Instead of signed short need:
int32_t CalculateAltitude(unsigned int pressure)

eshifri, 2022-05-17:
Good catch!

Ktr128, 2022-05-17:
Thanks Eshifri!
Thanks to the dialogue with you, I went on the right path...

Outwide, 2022-05-22:
I ask will this be fix for the NV14 be in a nightly soon.

Ktr128, 2022-05-24:
> I ask will this be fix for the NV14 be in a nightly soon.
I have no information when it will be included in the nightly build.
But if you need a firmware with a fixed problem, you can download it from here:

https://www.rcgroups.com/forums/showpost.php?p=49216907&postcount=4320

Outwide, 2022-05-24:
Hi thankyou so much for your reply.

I am new to all this and ask is there a group I may join, as I have some questions with regard to telemetry, it would help me better understand why something doesn’t work.

Again thankyou for your time.

https://www.rcgroups.com/forums/showpost.php?p=49216907&postcount=4320

eshifri, 2022-05-24:
Discord would be the best place to start: https://discord.gg/wF9wUKnZ6H

closed 2022-06-01 by pfeerick