EdgeTX/edgetx #1972: Select Model - Move

rhmaddox, 2022-05-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Select Model - Move
When trying to rearrange modems, you can only move them to different categories.  You can not  re-order them in there present category.

Using Jumper T16 with 2.7.1
nightly t16-49c585f.bin

### Expected Behavior

Either move to category, or re-arrange in current category 

### Steps To Reproduce

Push roller
Select Model
Move

From there you can only move to different category.   Not re-arrange in current category 

### Version

2.7.0

### Transmitter

Jumper T16

gagarinlg, 2023-03-18:
Now, with labels, things are different. 2.9 will bring additional changes. Can you test the model selection screen from the current nightlies, please. Maybe it now does work as you would like. There are different sorting options available.
make a backup of your SD card before testing!

mixMugz, 2023-04-10:
It's possible just to keep manual sorting options? I don't need to sort alphabetically and I don't like sorting by usage. I wish to sort manually all models by my decision, is that possible?