EdgeTX/edgetx #1978: Radiomaster TX16S MAX with FRM302 external module

Zabriel34, 2022-05-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am frustrated in matters of FLYSKY's customer Services! 
I am using Paladin  PL18 with its newest firmware (PL18, Frm301&receivers) and I love it. But I want to use some special functions wich PL18 is not supporting like some voice commands etc...
Thats why I recently bought the external module Frm302, to be able to use some of my AFHDS3 receivers with other open source radios. I updated the firmware of the module provided in Flysky's website and found on some other supporting open source communities, and flashed the appropriate firmwares to the radios (different versions of OpenTX and EdgeTX). 
But I was not able to bind this module to the recievers. It is also not binding in PPM mode in any of my other radios. 
I wrote an e-mail to Flysky, tried online support chat, asked about the orijinal firmware of the module. 
But I did not receive ANY answer yet! Also the seller of the module couldn't help me. 

Could this be a problem of EdgeTX firmware? I searched everywhere I could think of related to this problem. Tried everything I found in the Internet. But I could not find a solution for EdgeTX. 

Here is a video of my problem, for who are interested: 

[url]https://youtu.be/FDCL24ZkhT8[/url]

### Describe the solution you'd like

Any help is welcome!

rotorman, 2022-05-19:
Did you update your FRM302 to "OpenTX" compatible firmware? See: https://www.youtube.com/watch?v=p43xFniYZsg
Here the FRM302 firmware link from Paweł Spychalski's YouTube page: https://drive.google.com/file/d/1hoJQUjiR3jzayFHuM4HY7tEkoa-Wx_TX/view

Zabriel34, 2022-05-19:
Yes, I tried both of the firmwares. Also I tried the newest & some older versions of Open TX versions. Neither on OpenTX nor on EdgeTX the external module didn't bind. Also I tried PPM mode on both systems.

rotorman, 2022-05-19:
Does the FRM302 bind to the receiver with PL18 OK?

Zabriel34, 2022-05-20:
The receiver binds with the PL18. There is no problem. But I don't have the JR adapter (FJR2) for the PL18. Also, if I had it, Flysky is claiming if the firmware of FRM302 is changed to "developers firmware", the PL18 is not compatible with FRM302 anymore!

I watched the video you pointed and many more videos on Youtube, and tried really everyone of them. All the same result. 

But the most striking thing is, it is not binding in PPM mode either! I tried it in 2 other transmitters. Maybe because I changed the original firmware of the FRM302!? I cannot find the original file to rollback the FRM302. FlySky and the seller are not answering! Rollback the fimware to the original and trying it in PPM mode again is the only thing I couldn't try yet! If I can try this with no success, then I believe the module is faulty...

You are right, I posted this issue in RCGroups and in RCUniverse also. Because I want to reach OpenTX users and non-open source users who are using AFHDS3 external modules, beside EdgeTX users. Maybe out there some FlySky employee read it and answer me... 
Until now you are the only one who replied. 
I never give up hope! 
Thanks anyway.

Zabriel34, 2022-05-23:
No solution until this moment, I give up.

Zabriel34, 2022-05-31:
Hi there,

Zabriel34, 2022-05-31:
Hi there,
Finally I managed to bind the FRM302 module in PPM mode with "developers firmware" on it. I downgraded the receivers (FTr10) firmware, Then it was able to bind in PPM mode. But it will not work with the internal module in "Radiomaster Tx16S MAX" with EgdeTx 2.7.

rotorman, 2022-05-31:
Does it work with OpenTX?

Zabriel34, 2022-05-31:
No, BUT I was not able to test it with OpenTX 2.3.10. I tried it with 2.3.11 and 2.3.14 and 2.3.15. 

I recieved an e-mail from FlySky Tech Service. Here is the history: (I didn't understand it fully. It was more or less trial& error to me...) 

"Hi, my PL18 has the firmware 1.0.65. And my Frm301 and all the receivers are binding. They are working fine. But I didn't fully understand: 
1) If I downgrade just the receivers, do I have to downgrade the PL18 too? 
2) Will then bind Frm302 with "developers firmware", bind then to the downgraded receiver (FTr10) on a open source radio (like OpenTX or Edge TX)? Or do I have to use PPM mode with other transmitters? 
3) I would like to flash the original Firmware of Frm302, but I was not able to find the original file. Can you provide me just the original firmware of Frm302? 
...

Hi Deniz:
Thank you for your support of flysky.
First of all, you need to confirm your FTR10 and FTR16S, you can complete BIND with PL18+FRM301, which version is your PL18 at this time?
If your PL18 is version 1.0.49, then you can complete BIND with FRM302 without downgrading the receiver.
And if your PL18 is version 1.0.55 or 1.0.65, then you need to downgrade the receiver.
As for how to downgrade the receiver, you can download the receiver downgrade software on the flysky official website.
If you have any other questions, please feel free to contact us.
greeting
cheng

... 

"Dear Deniz:
I'm very sorry for the inconvenience caused to you.
First let me explain to you the following:
--PL18 has used the new high-frequency 3.0 program since 1.0.55, while the old high-frequency 2.0 program has been used in versions 1.0.49 or earlier;
--If the receiver wants to complete BIND with the transmitter, it must ensure that the transmitter, the high-frequency module and the receiver use the same high-frequency program, otherwise the BIND cannot be completed;
--If the receiver and PL18 1.0.49 or earlier versions have completed BIND, the HF 2.0 program must be used, and if the receiver and PL18 1.0.55 or later versions have completed BIND, the HF 3.0 program must be used;
--FRM302 currently has no updated version, that is, the high-frequency programs included in the firmware of 302 are all 2.0.

_**1) If I downgrade just the receivers, do I have to downgrade the PL18 too?**__ -- must ensure that the receiver, HF module (302+301), PL18 use the same HF program
**2) _Will then bind Frm302 with "developers firmware", bind then to the downgraded receiver (FTr10) on a open source radio (like OpenTX or Edge TX)? Or do I have to use PPM mode with other transmitters?**_ --OPENTX and EDGETX can't update the flysky receiver, the flysky receiver can only be updated with the flysky transmitter. And the developer firmware can only adjust the power.
**3) I would like to flash the original Firmware of Frm302, but I was not able to find the original file. Can you provide me just the original firmware of Frm302?**--You can get more information on FLYSKY official website and download related Manuals, view product parameters, compatibility tables for transmitters and receivers, software version update records, firmware, and more.
https://www.flysky-cn.com/0enhome
greeting
cheng"

richardclli, 2022-11-18:
I noticed that OpenTX 2.3.15 can use FRM302 without any problem.  You may check with this version first.
I am getting a FRM302 from Flysky and ensure it works with EdgeTX.

raphaelcoeffic, 2022-11-18:
@richardclli It won't work on 2.8. I warned everyone that supporting the new version of the protocol to the module of the EL18 would mean we could not properly support the FRM302, and that this module would need a FW update (FW not avail) to align it with the proto used by the EL18. We cannot possibly support 2 different version of the same protocol that requires different state machines.

raphaelcoeffic, 2022-11-18:
That being said, if the firmware of the FRM302 is aligned to the way the EL18 internal module works, then everything will be fine.

Zabriel34, 2022-11-22:
I cannot follow and understand the developments. I am not familiar with programming. 
I am now using FRM302 module happily with 2.7.1 on my Radiomaster TX16S MAX. 
As I understand I will not be able to use it with 2.8 ?! Therefore I am a little dissapointed.

pfeerick, 2022-11-22:
As stated above, it will require an update on Flysky's side, as they have implemented the interface with AFHDS3 differently in the EL18, and it is not compatible with the FRM302. Once that is done it should work again.

richardclli, 2022-11-24:
Well, as I know Flysky do not have any plan to update the firmware in FRM302 to support the new protocol, so you may need to stay with 2.7.1.

gagarinlg, 2025-01-24:
this will not be fixed FRM302 is unsupported by us and FlySky

closed 2025-01-24 by gagarinlg