EdgeTX/edgetx #1983: Companion 2.7.1 Simulator and Standalone Simulator crashes with this Model

ParkerEde, 2022-05-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

The EdgeTx 2.7.1 companion simulator crashes, after loading this OTX file in Companion and launching the simulator.
Also if I copy the MODEL and RADIO folder to the SD-Card Path, the standalone simulator (SD Card Mode) crashes after converting the bin file.

[EDGE-Simulator-Crashes_with_this_Model.otx.zip](https://github.com/EdgeTX/edgetx/files/8747460/EDGE-Simulator-Crashes_with_this_Model.otx.zip)
.

Please rename the file to EDGE-Simulator-Crashes_with_this_Model.otx

I inspected all settings in the modell programming windows, for me it looks fine. But anything went wrong.

### Expected Behavior

EdgeTx Companion will crash

### Steps To Reproduce

1. Start EdgeTX Companion 2.7.1
2. Open the attached OTX file
3. Start the simulation
4. EdgeTx Companion will crash

### Version

2.7.0

### Transmitter

FrSky X10 Express / X10S Express (ACCESS)

elecpower, 2022-05-21:
Does not crash for me however I do not have your sd card contents. There are known issues with themes causing crashes. A known issue is a single dash '-' for a tag value in the theme.yml file.

ParkerEde, 2022-05-21:
for the case:

_1.Start EdgeTX Companion 2.7.1
2.Open the attached OTX file
3.Start the simulation
4. EdgeTx Companion will crash_

I've also tried with any empty SD-Card folder and crashes too.

ParkerEde, 2022-05-21:
I've tested a clean installation on a clean windows 11 System, create a Radioprofile Horus X10S Express leave all other defaults and empty SDCard folder.
Then
_1.Start EdgeTX Companion 2.7.1
2.Open the attached OTX file
3.Start the simulation
4. EdgeTx Companion will crash_

and crashes again.

elecpower, 2022-05-21:
Tested on Win 10 and it crashed

ParkerEde, 2022-05-21:
Yes. It seems to be an issue on Windows Plattforms. Same test on an Ubuntu 20.04 works without crash.

elecpower, 2022-05-21:
Win 10 simulating new file and new model does not crash

elecpower, 2022-05-21:
Does the otx convert and work on the radio hardware?
Can you zip and upload the MODELS and RADIO folders from the hardware radio sd card after conversion so I can compare to what Companion is generating.

elecpower, 2022-05-22:
Win 10 simulating FrSky X12S from Companion and it crashes

ParkerEde, 2022-05-22:
I will test it this evening. Actually I run OTX on the Horus X10S Express. 
I will report.....

ParkerEde, 2022-05-22:
Now I've tested on Horus X10S Express. It works as fine as in Ubuntu.

elecpower, 2022-05-23:
Would be good if a Mac user could test.

pfeerick, 2022-06-08:
I was able to reproduce the crash on win 11 x64 in simulator (both via companion and standalone), but not via `simu` (although that is running via WSL2 so Linux build. Nothing useful in the debug logs as far as I could tell. I suspect it was something the simulator firmware didn't like in the model settings, as it loaded as much as the initial screen for the model then crashed. There may be some connection to #1933 as that also seems like a windows specific simulator crash.

ParkerEde, 2022-07-26:
Now I was able to check and reproduce the crash on Windows 11 Companion Simulator.
I used the attached Model file.
If I set in Companion GVARS 1-3 in FM0 to 100% than the Simulator will run. But if you open the "Global Variables" programming page in Simulator and change the value FM0 GVAR1 from 100 to 99 the Simulator will crash instandly.

Hope that helps to find the bug.

ParkerEde, 2022-09-09:
@pfeerick in 2.8.0 selfbuild there is no crash anymore. What should we do with this ticket?

pfeerick, 2022-09-09:
If it no longer crashes, If it no longer crashes, that hopefully means it was fixed (either expressly or accidentally). Either way, it no longer crashes for me either with a recent github build. I think it can be safely closed, and can be revisited if it pops it's ugly head up again. Normally you'd be fine to close it since you opened the issue, but it's good you pre-warned me as I'll tag it against 2.8 ;)

closed 2022-09-09 by pfeerick