EdgeTX/edgetx #1987: Radioking TX18S Not Working

efeksk, 2022-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Other (Please specify below)

### Current Behavior

Edgetx software was installed on my radioking remote. I wanted to convert it back to opentx again. After doing the bootloader process, the remote no longer works.

https://youtube.com/shorts/sMTxSDGS3tw?feature=share

### Expected Behavior

.

### Steps To Reproduce

..

### Version

2.7.0

### Transmitter

Jumper T18

### Anything else?

Edgetx software was installed on my radioking remote. I wanted to convert it back to opentx again. After doing the bootloader process, the remote no longer works.

https://youtube.com/shorts/sMTxSDGS3tw?feature=share

eshifri, 2022-05-22:
Can you please clarify what the "bootloader process" is?
Did yo use OTX SD after converting the radio back to OTX?
Can you still enter the bootloader?

norulers, 2022-05-23:
Insert USB in the shutdown state, select the correct target and refresh again

efeksk, 2022-05-23:
> Lütfen "önyükleyici işleminin" ne olduğunu açıklayabilir misiniz? Telsizi tekrar OTX'e dönüştürdükten sonra OTX SD kullandınız mı? Hala bootloader'a girebilir misin?

Hello. I can't get to the bootloader screen because the radio is not working at all. I made a mistake. I loaded opentx firmware belonging to Taranis x9d into Radioking TX18S remote. That's why the remote no longer works.

efeksk, 2022-05-23:
> Kapatma durumunda USB'yi takın, doğru hedefi seçin ve tekrar yenileyin

Hello. I can't get to the bootloader screen because the radio is not working at all. I made a mistake. I loaded opentx firmware belonging to Taranis x9d into Radioking TX18S remote. That's why the remote no longer works.

eshifri, 2022-05-23:
Aha. In this case your best (only) option is to re-flash it in DFU mode.
Connect the unpowered radio to computer with USB and use OTX companion to install the OTX.

efeksk, 2022-05-23:
When I connect the remote to my computer, I can see it as the stm32 bootloader. So how do I install the software? Should I open the inside of the controller and connect the usb programmer to the solder work on the processor legs? I do not know how to do it. I have no idea about this.

efeksk, 2022-05-23:
> Aha. Bu durumda en iyi (tek) seçeneğiniz, onu DFU modunda yeniden flaşlamaktır. Güç verilmeyen radyoyu USB ile bilgisayara bağlayın ve OTX'i kurmak için OTX yardımcısını kullanın.

I tried to connect the remote to the programmer via usb and upload OTX via compilation. But it was not successful

eshifri, 2022-05-23:
Start Companion.
Connect the TX without switching it ON.
In the Companion choose "Write FW to Radio".

efeksk, 2022-05-23:
> Companion'ı başlatın. TX'i AÇIK konuma getirmeden bağlayın. Companion'da "FW'yi Radyoya Yaz" seçeneğini seçin.

I tried to connect the remote to the computer via usb and upload OTX via compilation. But it was not successful

efeksk, 2022-05-23:
> Companion'ı kullandı. TX'i Açık eve getirmeden. Companion'da "FW'yi Radyoya Yaz" seçeneğini seçin.

I did this. But it didn't work.

eshifri, 2022-05-23:
Hmmmm... What was the error message?

efeksk, 2022-05-23:
> Hmmmm... Hata mesajı neydi?

The bootloader operation was not successful. It was a message similar to this

closed 2022-08-19 by pfeerick