EdgeTX/edgetx #1988: firmware flash on TX12 lost control of Betaflight

cristianrepo, 2022-05-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have download and self build  firmware(2.8.0, git 49c585f6)  and flash onto my TX12 Radio,  and found that I easily lost control of my quad drone 
during to Radio signals(drop),  I have tried flash back OpenTX with same hardware and there is no problem.
I saw the internal/external module driver codes have changed a lot from OpenTX, which or when there will be stable version?

Thanks

  

### Expected Behavior

stable version

### Steps To Reproduce

the firmware version shows: 2.8.0, git 49c585f6
Radiomaster TX12
betaflight 4.2.x

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX12

### Anything else?

2.8.0

robustini, 2022-05-26:
Same with Horus X12S and X10S Express.

raphaelcoeffic, 2022-05-26:
We will need a lot more infos ;-) for a start, which radio link is that? MPM? Which protocol? What kind of receiver? Is there any other symptoms? How do you notice? Is telemetry still ok? Anything else you can think of?

raphaelcoeffic, 2022-05-26:
> Same with Horus X12S and X10S Express.

D16 in your case? What happens exactly?

raphaelcoeffic, 2022-05-26:
Also, please note that 2.8 is a **development** version. You should use **2.7.1** for a stable version.

cristianrepo, 2022-05-26:
@raphaelcoeffic, thank you for your concern.  I would like to provide as more info. as possible. 
for my case: 
I was using inter-module within TX12, which is based on "Multi DIY" protocol I think.  I was using Frsky D16, no telemetry,  with a Frsky XM+ recerver.
I re-flushed 2.7.1 and the problem is the same.
I re-flush openTX and there is no problem with all same hardware.
What I noticed is like: the quad-drone flies and suddenly you can feel your RADIO lost control of the drone then the drone dropped onto the ground. Based on years of my experience, I think it is because the drone lost connection with RX...
Will betaflight log file be helpful? or tell me do some test if I can.

daleckystepan, 2022-05-26:
@cristianrepo Yes. Please post also BF log. In the best case BF log from BF4.3 the latest build and debug_mode=RX_TIMING.

Please also let us know firmware version of your internal module. Or update to the latest 1.3.3.7: https://downloads.multi-module.org

cristianrepo, 2022-05-29:
Please check the log files first. 
TX12 internal radio module should be  a "cc2500 only MULTI",  it should use same protocol as 4in1 MULTI, but can not support all 4in1 protocols
I have downloaded log viewer and saw at end there are always FAILSAFE...
I will switch to use external 4in1 MULTI where days. 
@daleckystepan @raphaelcoeffic
[LOGS.zip](https://github.com/EdgeTX/edgetx/files/8792481/LOGS.zip)

daleckystepan, 2022-05-29:
It looks like normal failsafe because of signal loss. However I can see some gaps in BF logs that can show troubles with CPU usage.

You have pretty nonstandard setup. Please update to the latest nightly build of Betaflight to get better logs. Use this setup on your F4 if possible:
* 8k/4k gyro loop/pid loop (you have currently 8k/4k)
* DSHOT 300 (you have OneShot125)
* debug_mode=RX_TIMING (currently RX_SIGNAL_LOSS)

Please consider updating your internal module and also your XM+ receiver:
* 1.3.3.7: from https://downloads.multi-module.org/
* ACCST D16 v2.1.2: https://www.frsky-rc.com/xm-plus-mini-sbus-non-telemetry-full-range/

Then I would also recommend to switch to FrSky X2 mode. D modes have signal quality issues: https://github.com/betaflight/betaflight/issues/11560

cristianrepo, 2022-05-29:
I have flash Betaflight 4.3 RC6  the latest  firmware.
I am using a very old ESC which do not support DSHOT300
I have change log  to 2K frequency, but I can not find RX_TIMING in Beteflight -  configurator - Blackbox pulldown  menu. @daleckystepan 
I will switch to use external mudole that is 4in1 Multi. Will do these tomorrow.

daleckystepan, 2022-05-29:
@cristianrepo You also need the latest Configurator: https://github.com/betaflight/betaflight-configurator-nightlies/releases/ but you should be able to see that debug mode. You can try it using CLI.

Ok. No worries with DSHOT. It should not cause anything like that. It was just advice.

cristianrepo, 2022-05-30:
Attached is today's log file. BF is 4.3, radio module is external 4in1.  
what I saw was that the drone only can take off the last time, and dropped about 20 seconds.  
strange I can not see FAILSAVE at end of of the last log file.
I am thinking what to do next, maybe change to another drone to compare. but, I want to use same receiver first. 
[LOGS2.zip](https://github.com/EdgeTX/edgetx/files/8796420/LOGS2.zip)

daleckystepan, 2022-05-31:
@cristianrepo I can see one common pattern among flights. Once you raise the throttle it failsafes.

Can it be some connection issue? Mechanical issues? Noise from motors? Loose antenna or wires?

[image]

cristianrepo, 2022-05-31:
@daleckystepan you are right, that's what I saw during the test!
I will switch to use another Quad to compare the resultl.

cristianrepo, 2022-06-02:
@daleckystepan ,  please check log file: 
1, same TX12 no any change
2, Quad is a new one: HAKRCF722, Frsky X4RSB receiver, D16, 32bits ESC DShot300
3, log at least 3 times, all the 3 times quad drop at last.
What I suggest is lets test an  older firmware, for example, a version before edgeTX heavily 
modify interface with radio module, or, before change RTOS.

daleckystepan, 2022-06-02:
@cristianrepo Thank you for your testing and thank you for proper debug mode. As you can see `Frame delta` should be around 9ms. Sometimes (or better to say very often) it is more than 50ms or even 70ms. RC link is not working and it is not safe to fly with this stuff until you will be successful and resolve the issue.

If I understand you correctly you have changed only the quad. TX remains the same and problem still persists. So there is chance that the issue is in your TX itself or TX module. Right?

Updating to different settings of BF or even to the newest BF hasn't changed anything.

[image]

cristianrepo, 2022-06-02:
@daleckystepan, thank you for your comments. Yes, I am using a new quad, with RX the same.
Before I submit this issue, I have compared edgtTX and OpenTX,  when I use openTX there is no such problem.
What I am planning to do is to test edgeTX version 2.50, I see it is a very early version, still using CoOS at that time.

cristianrepo, 2022-06-04:
@daleckystepan, 2 log zip files.
log4 is using TX12 edgeTX2.50, there is still drops. readme.txt inside
log5 is using Frsky X9LITE edgeTX271, buildin module, no drops.
I see there is RX link interruptions in log4, none in log5.
I suggest still focus on TX12, Multi module interface, more compare tests. maybe there is some place need to be improved.

[log4.zip](https://github.com/EdgeTX/edgetx/files/8836920/log4.zip)
[log5.zip](https://github.com/EdgeTX/edgetx/files/8836922/log5.zip)

daleckystepan, 2022-06-04:
Personally I don't see much difference between `log4` and `log5`. Of course `log4` failsafed but there are RC link issue even in `log5`. Maybe your X9Lite has better antenna or more powerful module so it is not pronounced to failsafe.

ETX version or BF version does not change anything, probably changing TX itself does not change anything.

As my previous suggestions:
* Check antennas
* Check all wirings
* Update everything to the latest versions. Don't forget to RX firmware and TX module firmware
* D16 is not much reliable in BF according to our testing. Try to use X2
* Or even the best you can do: Don't use FrSky RC links.

cristianrepo, 2022-06-07:
@daleckystepan,  you are right,  maybe the problem is because of the quad and wiring:
1, because +5V onboard is broken I was actually using external 5v to BF board.(seems working, but may cause problem)
2, to easily change different receiver, I used 3PIN terminal pairs to connect RX receiver, wire is long.
I have  changed a new OMNIBUSF4SD and re-wired receiver, there is no problem last time fly. log files attached.
This Issue closed as solved, edgeTX is so far perfect.   @daleckystepan, I would like help to do more test works if later version needed. 
[log7.zip](https://github.com/EdgeTX/edgetx/files/8849289/log7.zip)

closed 2022-06-07 by cristianrepo