EdgeTX/edgetx #1989: Min/Max Highlight Incorrect on Outputs Screen When "Inverted" is Selected

jmxp69, 2022-05-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

2.7.1 TX16s On the outputs screen when the control is moved there is a min/max highlight depending on the movement of the stick. When inverted is not selected as an option and the stick is moved, there is a discrepancy between the highlight and the numeric field that actually changes the output.

With no inversion:
Elevator pushed forward/up MAX is highlighted, and the output value assigned to MAX limits output.

With inversion:
Elevator pushed forward/up MIN is highlighted, but the output value assigned to MAX limits output.

### Expected Behavior

The highlight should highlight the field that will actually change the output when inversion is selected.

### Steps To Reproduce

1. Open Outputs Tab
2. Select an output governed by a stick with INVERT turned off.
3. Move the stick, make note of the highlight field, the numeric field, and the output value in the monitor.
4. Change the output value on the same line as the highlight and notice the change in output on the monitor. This is the expected behavior.
5. INVERT the output
6. Change the output value on the same line as the highlight and notice the output on the monitor does not change. 
7. Change the output value on the line not highlighted and notice the output on the monitor changes. This is confusing. The value highlighted should change the output.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

2.7.1 from github.

philmoz, 2023-10-06:
Is this still an issue - can't reproduce in 2.8.4 or later.

pfeerick, 2024-03-04:
@jmxp69 Can you review this and see if it's still an issue? Thanks ;)

PR #5367 (2024-08-05): fix(color): min/max highlight in output list swapped if inverted
Fixes #1989 

Use code from output edit page to highlight min/max values on output list page.
Fixes issue when output is inverted.
closed 2024-08-05 by pfeerick