EdgeTX/edgetx #1992: Express LRS LUA V2 does not work together with Yaapu  Widget

robustini, 2022-05-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Unfortunately I have found that Express LRS LUA script V2 with the Yaapu Widget in the background does not want to detect the data from the module, remains in "loading...".
If I remove the Yaapu Widget the ELRS LUA it worked as it should.
In OpenTX this doesn't happen, they both work, without any problem, so it's something related to EdgeTX.

### Expected Behavior

It should load all data from the R9M module as it does in OpenTX.

### Steps To Reproduce

Run the LUA script with Yaapu Widget active, it will remain in "Loading ...".
Shortly thereafter the script will fail as the image I post.
Tested with EdgeTX V2.7.1 on FrSky Horus X10S Express and Horus X12S (with ISRM), same exact problem.

### Version

Other (Please specify below)

### Transmitter

FrSky X10 Express / X10S Express (ACCESS), FrSky X12

### Anything else?

[image]

janrunejohansen, 2022-05-28:
Its working fine for me after migrating from OpenTX. I also have the R9M module(ELRS) and did just copy over the scripts, images and sounds from the OpenTX sdcard. Everything was working at boot and didnt have do anything except setup the widgets.

dkemxr, 2022-05-28:
The Crossfire Agent Lite script doesn't work correctly either with EdgeTx if the Yaaapu script is running. And perhaps some other issues as a result with odd delays.

raphaelcoeffic, 2022-06-13:
I think the widgets should not be run while standalone LUA is running.

robustini, 2022-06-25:
> I think the widgets should not be run while standalone LUA is running (to be fixed).

I agree mate, thanks! There's fix in progress about that?

raphaelcoeffic, 2022-06-25:
> There's fix in progress about that?

Not right now, but I think this is something that should be fixed before 2.8 gets released, yes.

robustini, 2022-06-26:
> > There's fix in progress about that?
> 
> Not right now, but I think this is something that should be fixed before 2.8 gets released, yes.

Thanks mate, so unfortunately at the moment I stay on OpenTX because this problem is relevant to me.
I await a fix when possible and switch to EdgeTX definitively, thanks!

raphaelcoeffic, 2022-06-26:
@robustini I just checked in `main`, and it seems widgets don't run while on another screen than main, so I guess that should fix the issue, right?

robustini, 2022-06-26:
@raphaelcoeffic I honestly tell you that I have not then tried again with the current code, yes to solve the problem it is enough that the widget is stopped if you are not on main, but when i tried it didn't work.
In this context it is the CRSF access by the widget and the LUA that compromise the functioning.
In fact, just remove the CRFS option from the Yaapu widget and the LUA works also with the widget in background.
The problem is when they both try to access the same telemetry resource.
To solve this imho it would be necessary to block access to the telemetry of widgets in case there is a running LUA.

raphaelcoeffic, 2022-06-26:
Yeah, we know it does not stop the widgets in 2.7.1. Was just saying, it will be fixed in 2.8, as it is already now in `main`.

robustini, 2022-06-26:
> Yeah, we know it does not stop the widgets in 2.7.1. Was just saying, it will be fixed in 2.8, as it is already now in `main`.

Ah well, I did not know, I try thanks!

robustini, 2022-06-26:
@raphaelcoeffic I confirm, it works, I close the issue, thanks!

closed 2022-06-26 by robustini