EdgeTX/edgetx #1995: Transmitter freee in midle of flight

Rivelino911, 2022-05-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The other day in 6th flight, my Tx entered in a Comma State. Screen was almost black, in background was a freeze  shutdown screen.
Tx is EachineTx16s, battery full.
Is 3rd day of using this firmware.
Seems is normal after power off twice. After first restart, entered again in this state.…

With other firmware's i haven't this problem.
Was a clean install, firmware/bootloader.

### Expected Behavior

To work normally.

### Steps To Reproduce

I can't reproduce it.

### Version

2.7.0

### Transmitter

Radiomaster TX16S, Other (Please specify below)

### Anything else?

Future firmware's can make a log?
Is better to find what is happening with radio.

MervDale, 2022-06-09:
I believe 2.7.1 has been a update to 2.7 might be worth an upgrade .

gagarinlg, 2022-09-22:
Are there any news on the issue?
Did you upgrade to 2.7.1?

Rivelino911, 2022-09-22:
> Are there any news on the issue? Did you upgrade to 2.7.1?

After that issue, i change it back to OTX. I don't need touch screen, just a reliable system to fly with. My heli's are very expensive. I don't make anymore test. That freeze cost me a lot, just totalized my back-up F3C competition heli. Thinking also to change radio Tx with one, who is not software SD card dependent. Like Futaba, or my old Taranis X9D+.

closed 2023-02-25 by gagarinlg