EdgeTX/edgetx #1996: Splashscreen can't be disabled

malinhiles, 2022-05-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

-On radio startup, the "EdgeTX" splash screen is shown for 2 seconds even though it is disabled in the radio settings.

-While the radio is displaying this splash screen (which isn't supposed to be there anyway cuz I turned it off in radio settings), the screen backlight is "deep blue" hue, and not the "white" like i have it set in the radio settings. After the radio has booted past the splash screen , the user defined "white" backlight color resumes.

I'm on edgetx v2.5.0 currently, but it the issue persists in v2.6.0 and v2.7.0 too. I flashed all versions and tested it. The newer firmwares were too buggy on my radio so I had to move back to 2.5.0.

### Expected Behavior

When the splash screen is disabled in radio settings, the splash screen doesn't show up

### Steps To Reproduce

1. Disable splash screen in "Radio Settings"
2. Turn on radio

### Version

2.5.0

### Transmitter

FrSky X9D+

Eldenroot, 2022-05-30:
What is buggy with 2.6.0 and 2.7.0? I use them without any issue. If you know about any bug/issue - please open a ticket here!

malinhiles, 2022-05-31:
> What is buggy with 2.6.0 and 2.7.0? I use them without any issue. If you know about any bug/issue - please open a ticket here!

Here's a couple others I experienced:
#1857 
#1858 

I've also had a handful of issues with the TBS Agent Lite Lua on edgetx v2.7.0. I didn't document the issues well enough for an Issue ticket though; just decided to move back to to v2.5.0 instead. 

Any thoughts on the Splashscreen issues?

sixer93, 2022-08-19:
Mine still splashes OpenTable but loads edge lol. I screwed something up lol

pfeerick, 2022-08-20:
If you're using a B&W radio, it means you're using OpenTX :-P You might have updated the SD card contents (including voice files) but not actually done the firmware flash? i.e. what does it say in Radio Settings/SYS -> Version? ;)

If it's a color-screen radio - I guess there could be a OpenTX splash screen in `/IMAGES` ?

pfeerick, 2024-01-08:
Duplicate of #1856

Is this still an issue? If you had issues starting with 2.6.0, given 2.6.0 onwards changed from settings being stored in EEPROM to being stored on the (Micro)SD card, there may have been a compatibility issue with your SD card, and it wasn't responding quickly enough during boot to read the initial boot settings. Plus, some startup related settings weren't being loaded properly on boot for a while on B&W radios.

malinhiles, 2024-01-10:
> Duplicate of #1856
>
> Is this still an issue? If you had issues starting with 2.6.0, given 2.6.0 onwards changed from settings being stored in EEPROM to being stored on the (Micro)SD card, there may [...]

Can no longer confirm if this is an issue; I sold my X9D 9 months ago. If I was the only one experiencing these issues, it very well could be the SD card I was using.

pfeerick, 2024-01-10:
Thanks for the reply. I'll close this now then as there have only been one or two reports, and the startup setting issues I know where present are gone, so up to someone else to report it again if there is still an issue. Happy flying! :)

closed 2024-01-10 by pfeerick