EdgeTX/edgetx #1997: Hott Telemetrie Values

Networker-001, 2022-05-27:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On EdgeTX 2.71 and Multiprot 1.3.3.7 some HOTT telemetry items have to be checked

-temperatur EAT1 and EAT2
always displaying -20 degrees celsius

-RSSI and TSSI
values are negativ ( fix in telemetry window with offset 120 works)
naming is not correct for Model Locator Rssi have to be changed into RSSI ( fix in telemetry window
zero value have to be treatet without offset

-Lua getrssi() does not work

pascal langer sees this issues not on the Multi Side

### Expected Behavior

Display correct temperatur values

Rename Rssi and Tssi

adding offset for values other then zero

check getrssi

### Steps To Reproduce

Using HOTT Receiver gr-18 and electric  air module

### Version

2.7.0

### Transmitter

Radiomaster TX16S

mha1, 2022-05-29:
Hi,

EAt1, EAt2: Are you sure you have working temperature sensors connected to the EAM? I don't have a physical EAM, just simulating it. Can you please send me a .bin log file generated with a Graupner HoTT TX.

Rssi, Tssi: are physical values estimating received rf power. dbm's are negative values. high values (e.g. -25dbm) are good, low values (e.g. -105 are bad) 

The confusion might be with the RSSI value which was historically used by FrSKY. FrSKY RSSI is not what it says. It's an estimate of the link quality (in OTX/EdgeTX HoTT terms Rlqi). Link quality values are ranging from 0..100 with 100 good, 0 bad.

Locator: the locator uses the old FrSKY convention for "rssi" which in reality is link quality. To make use of the locator as it is you need to look at Rlqi. To mimic RSSI add a calculated sensor in the model telemetry settings:

[image]

LUA getrssi(): I haven't tried yet but it should return the value of Rlqi which is a number from (0 to 100) plus the the two warning levels. How does it "not work"?

Networker-001, 2022-05-30:
Hallo Michael,

thank you very much for your very quick reply.

1.) Looking at the discovered telemetry items I found the item Rtmp showing the temperature.
This sensor ist part of the receiver. So sorry for this.

        2,) RSSI /  rssi
	I added a log file generated during a normal flight.
              You can see, that the value Rqly does not realy Change ist value, Rssi does

              I looked inside the Graupner manuel which says that the max value is – 20 dbm is the value for 100mW

              For myself I changed the LUA Script as follows
----------------------------------------------------
local function main(event)

  lcd.clear()
  
  local rssi = 0
  
  local rssi1 = getValue("Rssi")
 
  if (rssi1 == 0) then
    rssi = 0
  
  else 
    rssi = rssi1 + 120

  end  

              the model locator now works as aspected for Hott Receivers, showing 100 % at -20dbm

2.) Discussion

I think there are a lot of Graupner Hott  users that like to use edgetx but they are not the freaky guys to
change issues like this. It wood be good if the location finder is usable out of the box for them.

I see 2 possible ways for that:

1.) Change inside edgetx
2.) Deliver extra widget for Graupner hott

     So I‘am looking forward to hear from you

     If you have items to test with Hott Items like gr24, gr18,gr16, gr12 or general air module, feel free to ask.

Regards Martin
              

Von: Michael
Gesendet: Sonntag, 29. Mai 2022 17:30
An: EdgeTX/edgetx
Cc: Networker-001; Author
Betreff: Re: [EdgeTX/edgetx] Hott Telemetrie Values (Issue #1997)

Hi,
EAt1, EAt2: Are you sure you have working temperature sensors connected to the EAM? I don't have a physical EAM, just simulating it. Can you please send me a .bin log file generated with a Graupner HoTT TX.
Rssi, Tssi: are physical values estimating received rf power. dbm's are negative values. high values (e.g. -25dbm) are good, low values (e.g. -105 are bad)
The confusion might be with the RSSI value which was historically used by FrSKY. FrSKY RSSI is not what it says. It's an estimate of the link quality (in OTX/EdgeTX HoTT terms Rlqi). Link quality values are ranging from 0..100 with 100 good, 0 bad.
Locator: the locator uses the old FrSKY convention for "rssi" which in reality is link quality. To make use of the locator as it is you need to look at Rlqi. To mimic RSSI add a calculated sensor in the model telemetry settings:

LUA getrssi(): I haven't tried yet but it should return the value of Rlqi which is a number from (0 to 100) plus the the two warning levels. How does it "not work"?

mha1, 2022-05-30:
Hi,

1) I don't quite understand. Did you misinterpret EAt1 or EAt2 to show the receiver temperature? If so you are right to choose Rtmp, see naming conventions. The EA stands for EAM, R for receiver. So no problem here, right?

2) Rssi vs model locator RSSI.
As already explained the HoTT Rssi is a real signal strength indicator (therefore negative and with unit [db], while the FrSky RSSI is in reality an estimation of link quality, therefore ranging between 0..100, meaning 0..100%. Two different pairs of boots. The FrSky RSSI is a more alive estimation of link qulity than the Graupner HoTT one (here called Rlqy). Please also note that to calculate Rlqy Graupner receivers use different algorithms to estimate the link quality. All non-gyro RX's are more lively than the gyro receivers. Using e.g. a GR-18 (or GR-24pro or any other gyro receiver) will always show around 100% and only go down significantly when receptions goes really bad. Try for yourself and compare a GR-18 log to a GR-16 log. So really the locator should have used a true signal strength indicator in the first place, which FrSky probably doesn't offer as telemetry value. And they shouldn't have called it RSSI. The easiest way for your problem would really be having a separate Model Loacator for HoTT that is working with Rssi. I see what I can do to get this into the Multimodule script collection (https://downloads.multi-module.org/ - right at the end)  where you can also find the HoTT textmode LUA script and other Multimodule related scripts.

3) How about the problem you had with LUA getRSSI()? Works fine for me. For HoTT "value, low, critical = getRSSI()" delivers value = Rlqy (the uplink quality) and the low and critical warning thresholds, see [Lua doc](https://luadoc.edgetx.org/part_iii_-_opentx_lua_api_reference/general-functions-less-than-greater-than-luadoc-begin-general/getrssi)

Networker-001, 2022-05-30:
Hi,

1.  You are right. No Problem, i didn#t now the naming convention. Sensors Eat1 and 2 are not connected.
2. I’ve already have tried to post it to the multimodule, but they see it as an item for this board.
I‘ve checked my formula and changed it to

__________________________________________________________
  local rssi = 0
  
  local rssi1 = getValue("Rssi")

  if (rssi1 == 0) then
    rssi = 0
  
  else 
    rssi = (rssi1 + 80)/ 3 * 5

  end  
_____________________________________________________________
max value Rssi =    -20dbm  ->   100  db
min value Rssi =     -80dbm  ->        0 db

_____________________________________________________________

3. Also my failure I used getRssi() not getRSSI()

    Thank you verry mutch

   Martin

Von: Michael
Gesendet: Montag, 30. Mai 2022 19:09
An: EdgeTX/edgetx
Cc: Networker-001; Author
Betreff: Re: [EdgeTX/edgetx] Hott Telemetrie Values (Issue #1997)

Hi,
1. I don't quite understand. Did you misinterpret EAt1 or EAt2 to show the receiver temperature? If so you are right to choose Rtmp, see naming conventions. The EA stands for EAM, R for receiver. So no problem here, right?
2. Rssi vs model locator RSSI.
As already explained the HoTT Rssi is a real signal strength indicator (therefore negative and with unit [db], while the FrSky RSSI is in reality an estimation of link quality, therefore ranging between 0..100, meaning 0..100%. Two different pairs of boots. The FrSky RSSI is a more alive estimation of link qulity than the Graupner HoTT one (here called Rlqy). Please also note that to calculate Rlqy Graupner receivers use different algorithms to estimate the link quality. All non-gyro RX's are more lively than the gyro receivers. Using e.g. a GR-18 (or GR-24pro or any other gyro receiver) will always show around 100% and only go down significantly when receptions goes really bad. Try for yourself and compare a GR-18 log to a GR-16 log. So really the locator should have used a true signal strength indicator in the first place, which FrSky probably doesn't offer as telemetry value. And they shouldn't have called it RSSI. The easiest way for your problem would really be having a separate Model Loacator for HoTT that is working with Rssi. I see what I can do to get this into the Multimodule script collection (https://downloads.multi-module.org/ - right at the end) where you can also find the HoTT textmode LUA script and other Multimodule related scripts.
3. How about the problem you had with LUA getRSSI()? Works fine for me. For HoTT "value, low, critical = getRSSI()" delivers value = Rlqy (the uplink quality) and the low and critical warning thresholds, see Lua doc

mha1, 2022-05-30:
Alright. Pascal just rejected changing MPM code. He's right to reject it because there is no problem with MPM code. I'll try and ask him to include the attached Graupner HoTT Model Locator script in his Multimodule script .zip. This is were all the MPM protocol supporting scripts live and should be pulled from. Nothing to do with EdgeTX too. 

Here's the script. It is pulling sensor Rssi and scales values. I have seen around -15db when the TX beeing very close to the RX and HoTT receivers should have a sensitivity of around 110db. Scaling:

0db: 0%
 -1db to -14db: 100%
-15db to -115db  as 100% to 0%
< -115db: 0%

The screen says "db" but should really be %. The heck with it. Let me know if you think this is ok.

[Graupner HoTT Model Locator.zip](https://github.com/EdgeTX/edgetx/files/8800275/Graupner.HoTT.Model.Locator.zip)

Networker-001, 2022-05-31:
Hy Michael,

Looks good,

to my mind we have to look at the range of possible values and adapt it with a resolution factor.

If you look at the added script I used:

           local maxRssi = -15     (-20 is max Value from Graupner Manual)
           local minRssi = -85      ( I‘ve never seen -115)

and calculate the Resolution factor as:

            local RssiResolution =  - 100 / (minRssi - maxRssi)

also changed db in %:

  -- draw current value
  lcd.drawNumber(180, 30, rssi, XXLSIZE + CUSTOM_COLOR)
  lcd.drawText(260, 70, "%", 0 + CUSTOM_COLOR)

best regards

Martin

Von: Michael
Gesendet: Montag, 30. Mai 2022 20:57
An: EdgeTX/edgetx
Cc: Networker-001; Author
Betreff: Re: [EdgeTX/edgetx] Hott Telemetrie Values (Issue #1997)

Alright. Pascal just rejected changing MPM code. He's right to reject it because there is no problem with MPM code. I'll try and ask him to include the attached Graupner HoTT Model Locator script in his Multimodule script .zip. This is were all the MPM protocol supporting scripts live and should be pulled from. Nothing to do with EdgeTX too.
Here's the script. It is pulling sensor Rssi and scales values. I have seen around -15db when the TX beeing very close to the RX and HoTT receivers should have a sensitivity of around 110db. Scaling:
0db: 0%
-1db to -14db: 100%
-15db to -115db as 100% to 0%
< -115db: 0%
The screen says "db" but should really be %. The heck with it. Let me know if you think this is ok.
Graupner HoTT Model Locator.zip

mha1, 2022-05-31:
Sorry, but I can't get around calculating percentages on a physical value that has an inherent range for 0 to -infinity. Therefore I decided to display the real value in db and calculate a beep and color range along expected values.

Expected values are:
- 0dbm if no telemetry data is available (telemetry lost)
- high values up to -12dbm by experiment with a GR-24pro very close to the TX (try for yourself)
- low values down to -104dbm by checking the CC2500 transceiver data sheet (transceiver sensitivity) and I have logs with my big gliders at distances around 1000m at well below -90dbm

Using -115dbm to -15dbm for calculating the beep and color range is a reasonable compromise even if -115dbm is slightly under the theoretical limit of -104dbm. -15dbm to -115dbm is convenient for the scaling calculation as there is no multiplication or division necessary. The result is sufficient enough for the beep range, color range and the overall location task.

0dbm GR24pro turned off:
[image]

GR24pro in close proximity to TX:
[image]

[Graupner HoTT Model Locator.zip](https://github.com/EdgeTX/edgetx/files/8803126/Graupner.HoTT.Model.Locator.zip)

Networker-001, 2022-05-31:
Hi,

thank you Michael for the fast handling of this issue.

Martin

closed 2022-05-31 by Networker-001

mha1, 2022-06-07:
The script was accepted by Pascal to be part of the MultiLuaScripts package and will be available with the next MPM release (no date yet). Once the new MPM release is available please download **_MultiLuaScripts.zip_** from http://downloads.multi-module.org/.