EdgeTX/edgetx #1998: Betaflight LUA not getting data.

ewingate, 2022-05-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The Betaflight LUA script isn't loading data with my EdgeTx radio (Crossfire module for radio link). I have tried the release, and nightly build of the script. I am not running the Yaapu script. I have full telemetry coming from my flight controller without issue.

When I move the Crossfire module over to my Jumper T16 running OpenTx the same Betaflight script works as intended.

With the release version of the script I don't get past the "Loading VTX Tables", with the nightly it stalls at "Acquiring Device ID".

### Expected Behavior

Betaflight Lua script to populate with flight controller data when run.

### Steps To Reproduce

Load, compile, and run Betaflight LUA script.

### Version

2.7.0

### Transmitter

Radiomaster TX16S

pfeerick, 2022-06-03:
Have you checked the baudrate setting for CRSF in the model settings on the transmitter - IIRC 400k should be fine on TX16, 115k will probably work but would be slower.

pfeerick, 2022-06-03:
Wooho! Fantastic :)

closed 2022-06-03 by pfeerick