EdgeTX/edgetx #2000: Missing options in Custom Failsafe mode settings screen

mha1, 2022-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some protocols like FrSky and HoTT allow setting fail safe modes (Hold, Position, no Pulse) for individual channels. OpenTX allowed to set those under model settings, Failsafe mode, Custom, Set. EdgeTX doesn't have means of setting the modes. Here's what it looks like in OpenTX:

[image]

OpenTX user interface: long pressing return at a highlighted field toggles to the next mode (Position with position in %, hold, None for no pulse)

EdgeTX only allows changing of the position value. There is no means of selecting Hold or None.

 

### Expected Behavior

Allow selecting Hold and None as fail safe modes for individual channels.

### Steps To Reproduce

1. go to model settings
2. select internal Multi andHoTT or FrSky as protocol
3. select Failsafe mode Custom, press Set

### Version

2.7.0

### Transmitter

Radiomaster TX16S

raphaelcoeffic, 2022-06-15:
@mha1 i think we do have this page. I remember having to fix it a couple of releases ago.

mha1, 2022-06-15:
Then I need a little advice how to select hold and none. In OTX you longpress enter while being in the respective channel's field

raphaelcoeffic, 2022-06-20:
@mha1 Maybe I did not understand your question properly, but did you mean this?

[image]

The long-press on OTX is basically something to set the failsafe channels to the current values, right?
[image]

raphaelcoeffic, 2022-06-20:
@mha1 thx, now I see what you mean.

mha1, 2022-06-25:
@raphaelcoeffic  can you please take this out of triage

mha1, 2022-07-31:
@raphaelcoeffic  this seems to be fixed by the rework done to the Failsafe mode/Custom/Set page, right?

pfeerick, 2022-08-01:
It looks like it's mostly fixed... I think there is a bug of sorts around being able to set the failsafe values without a touchscreen... ie. when using the rotary encoder it scrolls between the parts of the number rather than changing the value... *somebody* forgot not all radios have touchscreens! :-P 😆 

[image]

https://user-images.githubusercontent.com/5500713/182052987-e5306543-8e2a-457a-8f40-0b3db70e570b.mp4

raphaelcoeffic, 2022-08-01:
@pfeerick it seems there is some bug preventing the `NumberEdit` event call-back to be installed properly. I'm on it.

raphaelcoeffic, 2022-08-01:
@pfeerick should be fixed now with EdgeTX/libopenui@fb01cf7ef9753e05d69

pfeerick, 2022-08-01:
Yup, that did it...  :)

closed 2022-08-01 by pfeerick