EdgeTX/edgetx #2002: Select model - Create model options to close.

Yannicflight, 2022-06-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

In the models menu, the popup that contains "Create model" is at the moment positioned right below the "Select model" button. This often causes me, and others around me to accidentaly create a new model instead of selecting the model I would actually want. When asking around on our model flying club I hear more people having this exact same issue.

This issue is applicable for both; using the touchscreen or the rotary select button, for the touchscreen the lines are just to close together / the buttons are to small. And for the rotary select button, I often rotate it by one click when pressing it which then also results into a new model being created. 

This often causes the accidental creation of a new model instead of selecting the model I would actually want. When asking around on our model flying club I hear more people having this exact same issue.
This does sometimes cause confusion since the model I want to control does not respond and you will end up with allot of unneccesary new models that you have to remove again.

### Describe the solution you'd like

The solution to this issue can be either:

- Moving the "Create model" button to another location can be an option but having "Duplicate model" or "Delete model" as the next option to "Select model" will cause the same kind of problem.
- Adding a secondondary confirmation to the create model button will be a solution so the process to create a new model requires 2 steps: Create model > Are you sure? Yes - No.
- Making the Select model button bigger or spaced out will already help when selecting it by touchscreen but will not help when accidentaly skipping a step when pressing the rotary button.

### Describe alternatives you've considered

- Pressing "enter" after accidentaly pressing the create model button to exit the "Model type" menu.
Doing this will stop the model type selection menu but it will create a new model anyway.

### Additional context

[image]
[image]

mha1, 2022-06-03:
check out this discussion: https://github.com/EdgeTX/edgetx/discussions/1883

philmoz, 2023-12-13:
Can this be closed?

pfeerick, 2023-12-13:
Yes, this is no longer relevant since 2.8.

closed 2023-12-13 by pfeerick

Yannicflight, 2023-12-13:
Yes this is solved, thanks!