EdgeTX/edgetx #2007: Ghost Menu issue

Martivip, 2022-06-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When I leave Ghost Menu from Edge TX TOOLS menu, the message "Remote menu" on my Ghost is still present it looks like after leaving Ghost Menu from Edge TX TOOLS menu the TX module stays in Remote menu mode.

### Expected Behavior

When I leave Ghost Menu from Edge TX TOOLS menu, the TX module schools also leave from that mode and the message "Remote menu" should disappear 

### Steps To Reproduce

Go to TOOLS menu on EDGE TX and launch Ghost Menu next exit from Ghost Menu by pressing the exit button.

### Version

Other (Please specify below)

### Transmitter

FrSky X9D+2019

### Anything else?

Edge TX version is 2.7.1

pfeerick, 2022-06-04:
This isn't a bug so much of EdgeTX, but of the user :-P ... If you long press, you force exit the tool, thus it is unable to send the "end session" type command to the ghost module. Just as happens with any Lua script, on both EdgeTX and OpenTX. 

Having said that, we have discussed that there should be a 'callback' facility for a script or tool to register a shutdown function for when it is exited, which would prevent this behaviour.

Martivip, 2022-06-04:
@pfeerick thans for the explanations if long press exit button is a force quit then indeed this is not a issue. Ok we can close the issue

closed 2022-06-04 by Martivip