EdgeTX/edgetx #2013: Failsafe Custom Settings page flickers

mha1, 2022-06-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Failsafe Custom Settings screen flickers after scrolling to higher channels
[image]

### Expected Behavior

No flickering screen

### Steps To Reproduce

1. Go to Model Settings, Model Setup
2. Scroll down to Failsafe mode and select Custom, the tap on Set
3. Swipe screen all the way down
4. Then swipe up again so that last line displayed shows CH13
5. Tap on the CH11 value 

### Version

2.7.0

### Transmitter

Radiomaster TX16S

### Anything else?

While you're at it no. 1: please check if positive failsafe values for Channels 9 to 16 are correctly coded to MPM format (for serial transmission to MPM)

While you're at it no. 2: please have a look at issue https://github.com/EdgeTX/edgetx/issues/2000

Thanks,
Michael

mha1, 2022-06-08:
Please also try this:
1. Go to Model Settings, Model Setup
2. Scroll down to Failsafe mode and select Custom, the tap on Set
4. Swipe screen all the way down
5. Tap on the CH16 value, then tap on the CH15 value

I think the entire page scrolling and indexing needs some attention.

_Re: While you're at it no. 1: please check if positive failsafe values for Channels 9 to 16 are correctly coded to MPM format (for serial transmission to MPM)_ 
I checked this myself using the MPM decoder and it looks ok.

raphaelcoeffic, 2022-06-15:
Not sure it is really fixing in 2.7, if we consider that 2.8 will basically fix all this (especially the visual keyboard).

mha1, 2022-06-15:
It is not only flickering, the indexing is faulty too. Just wanted to make sure this doesn't get overlooked at it is not only a presentation problem but also a functional one.

The functional part together with #2000 (if the problem is confirmed) might justify a 2.7 fix especially when 2.8 is pushed out as it touches a highly needed function for rc models: the ability to set fail safe values.

mha1, 2022-07-31:
@raphaelcoeffic this also seems fixed

pfeerick, 2022-08-01:
Looks all good. Thanks! :)

closed 2022-08-01 by pfeerick