EdgeTX/edgetx #2017: Pots / Slider warnings become opposite after file conversion bin to yml

MRC3742, 2022-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I believe this started after PR https://github.com/EdgeTX/edgetx/pull/1981

Use a model with pot/slider warnings S1, 6P, S2 not selected and LS, RS both selected and was originally saved as an .otx file.
TestModel1.otx file attached (remove .txt)

Opening that file in ETX Companion 2.71 and the selections will all stay the same. Saving to an .etx file and reopening will then have all items selected. (This part was fixed in the above listed PR)

Opening up that same file in the current ETX Companion selfbuild [67787ac] will have the opposite effect.
S1, 6P, S2 will be selected and LS, RS will be not selected. Saving to an .etx file and reopening will now keep these opposite selections intact.

Also converting on the radio from OTX model bin files to yml files will produce the same opposite results on radio GUI.

[TestModel1.otx.txt](https://github.com/EdgeTX/edgetx/files/8865908/TestModel1.otx.txt)

### Expected Behavior

Items converted remain the same on Radio and Companion

### Steps To Reproduce

1. Open TestModel1.otx in OTX Companion 2.3.15 and note pot/slider warnings selected.
2. Open that same .otx file in ETX Companion current selfbuild as listed above and selections will be opposite.

### Version

Other (Please specify below)

### Transmitter

Jumper T16

### Anything else?

The original otx file was saved from the Jumper T16 using OTX Companion 2.3.15 and then reopened as a Jumper T16 file in ETX Companion. So no different radio conversion involved just conversion to YAML ?

ETX Companion and Jumper T16 running Version 2.8.0 selfbuild (67787afc) 2022-06-08

elecpower, 2022-06-09:
confirmed

elecpower, 2022-06-10:
PR 1981 fixes issues caused by earlier PRs. All EdgeTX B&W radios are also affected similarly post PR 1981 and thus the need for a Release note in 2.8.

elecpower, 2022-11-16:
Closing as issue cannot be fixed as we are past the point of no return

closed 2022-11-16 by elecpower