EdgeTX/edgetx #2020: NO MULTI_TELEMETRY detected

SBWings, 2022-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I am not certain if this is an EdgeTX issue or a MPM issue. I was trying to help someone on Facebook resolve the following issue. Upon turning  on his RM TX16S, he went to his model settings. He scrolled to the Internal RF section. He had the message "NO MULTI_TELEMETRY detected". 

After exchanging several messages, I turned on my TX16S and scrolled to the same settings. I had the same "NO MULTI_TELEMETRY detected". I downloaded MPM v1.3.3.7 again and I re-flashed by going to the FIRMWARE folder on my SD card and selecting the correct .bin file. This did not help.

I scrolled up to the Internal RF/Mode field and I changed it from "MULTI" to "OFF". Because it's a computer, I rebooted my radio. After rebooting, I selected the MODEL SETUP page and I scrolled to the same field. I changed it back to "MULTI". I saw the progress bar titled "MPM: Scanning Protocols". Now it is working normally again.

This all seemed to occur when I created a new model or I duplicated an existing model with the FlySky2A protocol. I don't know if my radio was transmitting RF when this occurred. It wasn't bound to an RX at the time.

### Expected Behavior

To not see the message "NO MULTI_TELEMETRY detected" on all my models for unknown reasons.

### Steps To Reproduce

I don't know how I started receiving the message "NO MULTI_TELEMETRY detected" on the Internal RF section of my MODEL SETUP page. I The only thing I remember doing is duplicating a model with DSM protocol and changing it to FlySky2A with the PWM,IBUS option. I cannot reproduce the issue.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

bike797, 2022-07-07:
I have crazybeeF4FR board with spi frsky receiver and edge tx internal MPM module.
 I got same error messages sometime.

raphaelcoeffic, 2022-07-07:
This message appears anytime the radio does not receive data from the MPM module. Having a precise way to reproduce would be highly beneficial! Otherwise it is going to be quite hard to debug.

pfeerick, 2022-12-01:
Any further information on this? As indicated in the above comment, this error being shown is in relation to the link between the MPM RF module and the main radio board - not the RF telemetry link - so if you are getting that error, it is suggestive of an issue with the handset itself, given we haven't had any other reports of a similar error. 

It could be that the connector between the MPM board and the mainboard isn't plugged in properly. It will also show up if for some reason you enable the "No Telemetry" option.

SBWings, 2022-12-01:
I believe you can close this issue. It has not occurred since. I now believe it was operator error. I think it had to do with the selection to turn off the internal RF module in Radio Settings/Information/Modules / RX version/Internal module.

It's been several months so I didn't remember having this situation until your reply came into my Inbox that I remembered this. My Internal RF module was turned off under radio settings. If I recall correctly, once I set this to my internal 4-in-1, I went back to Model settings. I turned on the Internal RF module and it scanned in the normal protocols.

One final thought, I know I was attempting to use my radio as an HID device for RealFlight 9.5s. I might have turned off my Internal RF module to conserve battery life. I also sort of remember this happening around the time this setting became an option in EdgeTX. 

All is good now. Thanks for all your hard (and brilliant) work!

pfeerick, 2022-12-01:
Fantastic!  :) Yes, turning the module off in the hardware setup would indeed be a culprit also - better to use the model setup option and have a specific simulator profile. ;) Thanks for getting back so quickly... I was also just going through old messages and making sure none were missed.

TabajaraLabs, 2024-06-26:
I have the same issue and I was able to isolate the problem somehow. 
Just got an used Jumper T16Pro V2 Hall. This radio had OpenTX installed, Multimodule firmware 1.3.1.59. I updated it to the newest EdgeTX build avaiable (V2.10.1) and newest multimodule firmware avaiable (V.1.3.4.0). Got the "NO MULTI_TELEMETRY detected" message. 
To resume a long night of tests, my radio only works with the V1.3.1.59 multimodule firmware version. I used the old file which was still on sd card (still with opentx on its filename) and downloaded another 1.3.1.59 file from the site. If I load ANY file with version beyond 1.3.1.59, it will give the NO MULTI_TELEMETRY message. 
If I can be of any help to find this fault just page me. Thanks.