EdgeTX/edgetx #2021: Yaapu Lua script

ardisd, 2022-06-09:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There appears to be a bug running the yaapu Lua script or edgeTX 2.7.1. The widget keeps restarting on my RadioMaster 16TX MKII

### Expected Behavior

The Lua script should display telemetry data and not restart

### Steps To Reproduce

The issue happens when the script executes

### Version

2.7.1

### Transmitter

Radiomaster TX16S

athertop, 2022-06-10:
The yaapu widget config screen uses gVar 9 of FM8 to record a flag telling the widget if some config changed. If the widget sees gVar 9 of FM8 != 0 then it does the reinitialise you are seeing then tries to set the value back to 0 - but it fails to do any of this because of a recent commit in EdgeTx which affects how lua updates gVars. This is the commit that fixed a longstanding issue in OpenTx, but caused the issue for yaapu: https://github.com/EdgeTX/edgetx/commit/0391174e066a8b28cb87c656bec787a4a0fb648c
I have spoken with Alex (yaapu dev) and he is aware of this issue. I am sure he will address the issue inside the widget code. As this is not specifically an EdgeTx issue, it should not really be reported here.
I have also written a post all about this in RCG here: https://www.rcgroups.com/forums/showpost.php?p=49299729&postcount=852
This includes how to work around the issue - its a very simple config change you will need to do for each model that uses the yaapu widget.

AndreHenke, 2022-06-20:
Yaapu telemetry is not working on Edge TX.
I reinstalled yaapu on my TX16s, and screen goes black.
I tried to change the theme, but half of the information does not appear.
Should I go back to Open TX??
Thanks you!

rotorman, 2022-06-20:
Just tested, works fine.
Setup: RadioMaster TX16S mkII with EdgeTX v2.7.1 release.
Yaapu LUA telemetry widget (dev-branch)
ArduPilot with MAVLink telemetry
Mav2PT converter

https://www.youtube.com/watch?v=CTjDNwXNxrY

rotorman, 2022-06-21:
From YT video https://www.youtube.com/watch?v=CTjDNwXNxrY comment I take it, the issue for @AndreHenke is solved:
[image]

rotorman, 2022-06-21:
@AndreHenke I understood from your YouTube comment that you got it working already. 1.9.5 is not correct.
To git clone the dev branch: `git clone -b dev https://github.com/yaapu/FrskyTelemetryScript.git YaapuLUA`

AndreHenke, 2022-06-21:
I installed 1.96 as directed on the page and it is working, but not as good as it was on the previous version. Maybe there are some bugs that are being handled now and will be corrected soon. Will keep watching!
Thank you both!