EdgeTX/edgetx #2022: ACCESS receiver count value incorrect in model yml file

rburrow87, 2022-06-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With 1 receiver bound, the `receivers` value is 1. With 2 receivers it goes to 3, and with 3 receivers it jumps to 7.

### Expected Behavior

It should have a value of 1, 2, or 3.

### Steps To Reproduce

Register/bind 2 or 3 receivers and look at the `receivers` value under `moduleData` of the model yml file.

### Version

2.7.1

### Transmitter

FrSky X9D+2019

### Anything else?

I'm guessing this is not intended behavior, but if it is, #2019 will need to be adjusted to count receivers the same way.

1 receiver bound, `receivers: 1`
```
moduleData: 
   0:
      type: TYPE_ISRM_PXX2
      subType: ACCESS
      channelsStart: 0
      channelsCount: 8
      failsafeMode: NOPULSES
      mod: 
         pxx2: 
            receivers: 1
            racingMode: 0
            receiverName: 
               0:
                  val: "RX4R    "
```

2 receivers bound, `receivers: 3`
```
moduleData: 
   0:
      type: TYPE_ISRM_PXX2
      subType: ACCESS
      channelsStart: 0
      channelsCount: 8
      failsafeMode: NOPULSES
      mod: 
         pxx2: 
            receivers: 3
            racingMode: 0
            receiverName: 
               0:
                  val: "RX4R    "
               1:
                  val: "R4      "
```

3 receivers bound, `receivers: 7`
```
moduleData: 
   0:
      type: TYPE_ISRM_PXX2
      subType: ACCESS
      channelsStart: 0
      channelsCount: 8
      failsafeMode: NOPULSES
      mod: 
         pxx2: 
            receivers: 7
            racingMode: 0
            receiverName: 
               0:
                  val: "RX4R    "
               1:
                  val: "R4      "
               2:
                  val: "SpecV2 "
```

elecpower, 2022-06-10:
I haven't checked firmware but Companion interprets the receivers as a bit mapped field NOT a count.

rburrow87, 2022-06-10:
In retrospect, I'm not entirely sure this is actually an issue since it looks like both Companion and the radio firmware write out the same number. I expected it to be a count of 1-3 but it seems to be something else 😳

elecpower, 2022-06-10:
This is one of many yaml values requiring interpretation. As it is working as expected I'll close

closed 2022-06-10 by elecpower