EdgeTX/edgetx #2023: Servo Endpoint Adjustment Enhancement

jasc76, 2022-06-10:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When you have an existing model and smth changes like you adjust links or need to swap a servo or just because you need to tweak stuff, mixers and inputs and everything changing the channel values get in the way of properly adjusting servo endpoint. 

For me on mainly on multi flap gliders. Flaps are driven by aileron and crow break. Up/down is controlled differently and also there is the differential setting. So you cant just adjust endpoints
At least for my understanding, you need to edit mixers or create a new model whatever.

### Describe the solution you'd like

There should be a possibility to temporarily assign a stick, pot, slider to a output line to drive this output without any input, mixer or whatever. I guess there are more than one way. 
But what I would not do is, that you mark a line, choose the option and get a long list of possible inputs. Thats just too cumbersome. 
My proposal would be:
output Line, long press:
- (checkbox) temporary input ([show currently selected Input])
- set temporary input -> menu to select whatever stick, pot, slider you like

as long as "enable" is set, you drive this output only with the temporary input. 

### Describe alternatives you've considered

1. copy model and change mixer lines
2. create new mdel, same receiver with blank inputs and mixers

mha1, 2022-06-11:
I think Mike Shellim described this concept a long time ago. He called it calibration mode (https://rc-soar.com/opentx/setups/calmode/calmode.htm). It is a very useful concept for strongly separating the functional logic from the physical world (servos, linkages, endpoints, synchronizing servos, ...). He shows how this calibration concept can be easily implemented using mixers and flight modes. Works well.

raphaelcoeffic, 2022-06-11:
True, the setup is normally somewhat cumbersome but very useful. This needs quite some thought as to how it could be used in a useful way. Suggestions welcome!

mha1, 2022-06-11:
Basically it's decoupling the outputs from the mixers and feeding the output a virtual mixer with a fixed -100..+100 mix output driven by a source (e.g. S1). A mixer line with weight 100, offset 0, source S1 and mode REPLACE does exactly that. The major challenge is to make sure this can never be activated in normal operation. This is why Mike uses inhibit switches and a reserved flight mode that is set inactive for normal operation.

To implement a calibration mode I would:
- have a model defined selectable calibration source
- add a calibration check box added to each output line
- disable the check box(es) at any movement of any other control than the calibration source or leaving the output tab to prevent disaster.

User story:
- set e.g. S1 as calibration source in model settings
- select the output tab
- activate calibration of one or more channels (can not be activated without defined source)
- play with S1 and adjust min, subtrim, max to accommodate for mechanical limits etc.
- deselect check boxes or leave output tab or move any control other than S1 in this example

jasc76, 2022-06-12:
@mha1 

your user story sounds very good to what my thoughts were. Also a good idea to put it in the model setup. At the same time I think this setting should be kept logically within the same functional area, within outputs. There is already a button to set trim->subtrim so another function to set the calibration source would also fit.

only active within output tab is a nice addition.

yes it would be a „virtual“ mix line 100 weight, 0 offset, no trim

To deactivate the calibration if any other Input is moved, I wouldn’t do that. If you use maybe aileron as calibration source and you are working you can’t prevent some slight Input with elevator. 
If the function is deactivated when leaving output, this should be enough. Another option would be a warning if calibration is turned on on any line and you try to leave output.

mha1, 2022-06-12:
My reasoning behind "deactivate on any movement of any other control" is safety first. And the fact that for calibration (I use it the rc-soar way) it is not necessary to use sticks as source. Pots like S1, S2, RS, LS serve the purpose much better as you can set and hold any desired servo position without jiggling on a stick. Drive the servo near to the mechanical limit requires to move it with precision. You can't do that with holding a stick somewhere in between neutral and max. So really after experience with setting up my (up to 6 flaps) gliders I'd even restrict the possible sources to pots.

jasc76, 2022-06-12:
Fair
But as argument against, the TX16s has very unprecise center detents. That makes setting mid maybe unpredictable.

sure safety, but lets be honest: where is the risk? You activate and then you deactivate and go fly or whatever.
You can display a warning when someone leaves output page with calibration input activated. I think that’s sufficient and I would propose that. Or that this only works when output page is active.

There is so much people can screw up (like setting a wrong switch in mixers and disabling their channels in flight), I think this should not be overestimated

mha1, 2022-06-12:
Remember Murphy's law number 1? If it can go wrong it will.

Fair argument about unprecise centering on the pot sources (you could checke the source value that is displayed in the top right corner though) 

But I could live with deactivation on leaving the output page only and with this having sticks as sources possible.

Zaquron, 2022-08-25:
If i remember correctly MPX radios had this function since the Royal Evo 
Servo calibration was an absolute limit to the output. You selected the endpoint, longpressed a button and it moved the servo there. Then it could be adjusted with the rotary encoder.
This could be an endpoint button on the screen for each end and a long press would move the servo there until you quit that menu or move the sticks maybe?

jasc76, 2022-08-25:
Long time ago my buddy flew a Multiplex :-)
Sounds good, the only downside is your control surface slams to the endpoint and probably against the mechanical limit. So I would still prefer some control via some proportional input

Zaquron, 2022-08-29:
> Long time ago my buddy flew a Multiplex :-) Sounds good, the only downside is your control surface slams to the endpoint and probably against the mechanical limit. So I would still prefer some control via some proportional input

A popup could be added to alert the user to disconnect links or the starting point can be something reasonable like 30% of full travel maybe?