EdgeTX/edgetx #2024: Second decimal digit of sensor value is always zero

herrkami, 2022-06-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I'm using a Graupner GR-12L receiver with a HEPF micro vario module and the Jumper T-Lite. The micro vario detects sink rate, altitude, air pressure in hPa, and temperature. Pascal Langer's nice Graupner HOTT.lua script detects everything and I can see there that the altitude has a precision of 10 cm: 
[image]

However, in the telemetry tab of EdgeTX the sensor data is interpreted wrongly: 
1. The altitude is recognized with a precision of 1 m. Even when the precision is manually set to `0.0`, the second digit stays 0 while the first digit behaves as expected: when rising the sensor by one meter, it shows 0.0 m and then jumps to 1.0 m.
2. The air pressure is wrongly recognized or not at all (see screenshot).
3. The temperature does not show up at all. 

[image]
[image]
[image]
[image]

So overall, only three of the four sensors get recognized and only the sink rate is interpreted correctly. 

### Expected Behavior

1. Altitude should be readable at precision "0.0" or even "0.00".
2. Air pressure should be recognized and interpreted correctly. 
3. Temperature should be recognized and interpreted correctly.

### Steps To Reproduce

1. Connect HEPF uVario to Channel 5 input of the GR-12L and bind
2. Call the `Graupner HOTT.lua`-script and set channel 5 to sensor
3. On the other pages you should see the sensor data being received and displayed correctly
4. Exit the script, go to the telemetry page, and discover new sensors
5. In addition to the default receiver sensors three new sensors should pop up and behave as described above

### Version

2.7.0

### Transmitter

Jumper T-Lite

herrkami, 2022-06-10:
Hey @mha1, 
Thank you so much for your explanation! Okay, I see, I will close the issue. However, can you hint me to a comprehensive documentation of the HoTT-protocol? I found some notes in code comments about addresses and byte order but nothing on the transmitted precision or format for example.

closed 2022-06-10 by herrkami

mha1, 2022-06-11:
Unfortunately Graupner never made the specification public. Some people have it but signed NDA's for it, so they are not allowed to talk about it. This left more or less well documented reverse engineering efforts. A start can be to look at other peoples open source implementations or just google HoTT DIY sensor. Grabbing a logic analyzer to see what's going on on the telemetry bus also helps to understand.