EdgeTX/edgetx #2025: Trainer Channel Source mismatched

TheIsotopes, 2022-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Trainer Channel Source readout from receiver is mismatched on Companion.
On my TX i'm using AETR but Companion reads out / shows RETA.
After Write back to TX all is good.
Imo it should an bug on Companion side.

### Expected Behavior

Same Trainer Channel Source on TX and Companion.

### Steps To Reproduce

- use Trainer Channel Source AETR on Companion
- write this to TX
- compare output from TX and Companion

### Version

2.7.1

### Transmitter

Radiomaster TX16S

TheIsotopes, 2022-09-28:
Nothing has changed in the current v2.8.0-RC1 either. It's a pity!
However, this only seems to affect color RX.

pfeerick, 2022-09-28:
Well, you haven't bumped the issue at all to see if there was any comment either ;) 

So by "trainer channel source output" you mean this?

[image]

TheIsotopes, 2022-09-28:
Yes, exactly this.

pfeerick, 2022-09-28:
Ok, I can reproduce this, and from the looks of the values I think the radio firmware has been writing it out of order, and companion is correct. 

I'm using TAER  channel order. So my trainer setting should be 
[image] 
[image]

However, when I read the radio with Companion I get
[image]

And I have no idea atm how the heck these settings correspond with the radio settings...
[image]

since the values Companion wrote make more sense - 0(R) is CH3+1, 1(E) is CH2+1, 2(T) is CH0+1 and 3(A) is CH1+1
```yml
  mix:
    0: 
      srcChn: 3
      mode: REPL
      studWeight: 100
    1:
      srcChn: 2
      mode: REPL
      studWeight: 100
    2:
      srcChn: 0
      mode: REPL
      studWeight: 100
    3:
      srcChn: 1
      mode: REPL
      studWeight: 100
```

pfeerick, 2022-09-28:
Ok, as secondary aspect I had never noticed between B&W and colorlcd - "default channel order" ("rx channel ord" on smaller B&W screens) which changes order of inputs and mixes on new models also changes the order the labels are shown on the trainer page... i.e. trying TAER and AETR order on B&W will show trainer channels in that order... and on colorlcd it's not moving...

PR #2401 (2022-09-29): fix(color): Trainer stick labels not correct
Stick labels for trainer mode should match default channel order setting

Fixes #2025

Just need to sanity check/test again before mislabeling them completely! :-O However just referred back to otx and it looks right. 

https://github.com/opentx/opentx/blob/28f67c1e9a4e949c236f576a215e9becb52fd6ec/radio/src/gui/480x272/radio_trainer.cpp#L54
closed 2022-09-29 by pfeerick