EdgeTX/edgetx #2031: Can't exit "Modules / RX version" modal with return button, only touch

jurgelenas, 2022-06-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I can't exit "Modules / RX version" modal with the transmitter physical return button. Only by touch: 

[image]

### Expected Behavior

Modal "Modules / RX version" is closed after return button is pressed

### Steps To Reproduce

-

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-06-12:
Correction: you can exit without touch by using the Enter button since the exit button is in focus, but exiting via Return is indeed borked - although it is somewhat by design atm as the alternative was an emergency mode crash, which is less desirable.

rotorman, 2022-06-12:
The touch exit from this dialog "works" also if you click slightly to the left of exit button, but clearly the touch coordinates do not match the displayed Exit button coordinates.

gagarinlg, 2022-06-12:
In main exit via Return works, also via Enter
It das not work not with touching the EXIT button, but with touching outside of the window it works.

pfeerick, 2022-09-19:
Was fixed in #2035

closed 2022-09-19 by pfeerick