EdgeTX/edgetx #2032: RTC voltage decimal in wrong position

rburrow87, 2022-06-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

RTC voltage shows 29.8v

### Expected Behavior

Should be 2.98v

### Steps To Reproduce

Check RTC voltage in the radio settings -> hardware menu

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

[image]

Version 2.8 self build
[image]

raphaelcoeffic, 2022-06-12:
Please retry with current `Main`, was fixed this afternoon as well.

rburrow87, 2022-06-12:
This also looks good now!

closed 2022-06-12 by raphaelcoeffic