EdgeTX/edgetx #2033: Home screen widget margins incorrect

rburrow87, 2022-06-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The home screen widget margins are incorrect with various combinations of Flight mode/Sliders/Trims toggles.

### Expected Behavior

Widget margins should change appropriately when various combinations of Flight mode/Sliders/Trims are enabled.

### Steps To Reproduce

Have widgets on the home screen and toggle the different options in that view.

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

### Anything else?

Version 2.8 self build
[image]

Screenshots of settings and the result below.

1:
[image]
[image]

2:
[image]
[image]

3:
[image]
[image]

4:
[image]
[image]

5:
[image]
[image]

raphaelcoeffic, 2022-06-12:
@rburrow87 would you mind retrying with `main` again? I believe this has been fixed this afternoon.

rburrow87, 2022-06-12:
Looks good now!

closed 2022-06-12 by raphaelcoeffic