EdgeTX/edgetx #2037: Up/Down and Left/Right indicator on pages with more than one screen

SBWings, 2022-06-13:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When I'm on a page, there are some that are comprised of more than one screen, like the Model Settings page in v2.7.1. As a beginner, if I didn't try to scroll up or down, I wouldn't realize there were more settings beyond what I see on that initial page. This actually happened to me when I was first starting with either OpenTX or EdgeTX.

### Describe the solution you'd like

I'd like to see them to add some way of indicating there are more fields that are not shown on a particular screen, like a tiny up or down arrow ^ or three dots. In Android & Windows, scroll bars appear to let me know I'm not seeing all the available settings on my screen.

raphaelcoeffic, 2022-06-13:
Do you mean scroll bars? They need to be re-added, indeed :-)

SBWings, 2022-06-14:
Yes or something like "Page 1 of 3". Some OS's use an ellipse "..." to indicate that we the are more settings beyond what we can see.

closed 2022-12-06 by SBWings