EdgeTX/edgetx #2049: white screen emergency mode

pds59, 2022-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

radiomaster t16 s, load file t16s-f6988c0.bin, screen become white and emergency mode comes up  when you turn off.
after bootloader mode, load file, and radio look like is working ready for testing firmware

### Expected Behavior

no white screen, no emergency mode

### Steps To Reproduce

turn on radio 
connect to pc 
select usb storage
drag file to firmware folder and the screen become white
after complete remove usb cable
emergency mode show up
turn off radio
turn on radio and looks like is fine

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

gagarinlg, 2022-06-15:
Maybe you want to edit your last message. Having the cell phone number on the Internet might be a bad idea.

pds59, 2022-06-15:
Noted tks is due to my company auto reply

closed 2022-06-17 by raphaelcoeffic