EdgeTX/edgetx #2050: Number in trim bar (on change)

Eldenroot, 2022-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Select "change" option in "show number in trim bar" in MDL setup

(I use the Czech language, not sure if the translation is correct)

Try to change trims - no number is displayed on change... it is displayed only if you select "Yes" in the settings.

### Expected Behavior

Value should be visible

### Steps To Reproduce

Select "change" option in "show number in trim bar" in MDL setup

(I use the Czech language, not sure if the translation is correct)

Try to change trims - no number is displayed on change... it is displayed only if you select "Yes" in the settings.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

RJMcInty, 2022-06-28:
I've repro'd this on an x10S, running 2.7.1, running English language. It looks like:

- Display Trims = No : works as expected
- Display Trims = Yes : works as expected
- Display Trims = Change : seems to work the same as "No"

pfeerick, 2022-06-28:
This is also the case with yesterdays nightly on TX16S, which is the more important version to test right now as a lot of the UI code has been redone due to LVGL. 

Relevant code is here... and note the TODO ;) 

https://github.com/EdgeTX/edgetx/blob/ce8cc0b540dc287c41d52e1c601b37e762e8c35c/radio/src/gui/colorlcd/layouts/trims.cpp#L98-L111

RJMcInty, 2022-06-28:
Apologies if this is considered a "rude" question, but what's the user scenario for this feature (only temporarily displaying the value after a change)? I'm having trouble imagining a situation where I care about the number but only for a short while after a change, and after that it would bother me to have it displayed. I get when I may want to know the actual number, and I get that most of the time I might not care about the number. I don't see the middle ground though.

@pfeerick , you added a B&W tag, did you mean to add color instead (on simulator the B&W seems to work as expected)?

pfeerick, 2022-06-28:
I have no idea what you're talking about... it says color 😁 Thanks ;)

I think is like best of both... you can see the position all the time, and the number pops up on change for about two seconds if for some reason you need to sometimes know the value. On OTX it's like ETX B&W, the number is to the side of the knob... but that part isn't too important... 

[image]

RJMcInty, 2022-06-28:
Oh, I get how it works, I just don’t understand the “why” of the scenario. What’s an example where it would be useful for it to only show up for a few seconds, then go away?

raphaelcoeffic, 2022-06-28:
I was wondering as well…

Eldenroot, 2023-02-12:
Still valid, maybe lets discuss if there is a better way how to implement?

closed 2023-03-14 by pfeerick