EdgeTX/edgetx #2051: Model categories name is not displayed for empty lists

Eldenroot, 2022-06-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Slow name change

### Expected Behavior

Instant change on name change

### Steps To Reproduce

Create a new model category -> add an own name -> scroll into the new category -> the name is not displayed, it shows "new" for a long time... after some time it is changed.

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-06-16:
Please compare with a nightly if you can - the LVGL UI re-write was merged in the last few days so this may no longer be relevant.

Eldenroot, 2022-06-16:
Srry, I cannot try, I have only one radio :(

rotorman, 2022-06-16:
Well, basically you just need a backup of your SD card. Ideally get a second card for testing, then you are all set for wild testing with your one TX16S.

gagarinlg, 2022-09-22:
With the new model lables stuff merged, this issue is not relevant anymore.

closed 2022-09-22 by gagarinlg