EdgeTX/edgetx #2053: Pull OpenTX LUA ghostTelemetryPop() & ghostTelemetryPush() functions

JimB40, 2022-06-16:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No problem

### Describe the solution you'd like

This was introduced in OpenTx 2.3.15 and allows LUA scripts like BF Config to work properly wirh Ghost RCLink

raphaelcoeffic, 2022-06-21:
Already implemented with #1377

raphaelcoeffic, 2022-06-21:
And released since 2.7.0:
[image]

closed 2022-06-21 by raphaelcoeffic