EdgeTX/edgetx #2054: USB connection ends with Emergency Mode

jogybaer0815, 2022-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Radiomaster TX16S
nightly #ebcd61f (2022-06-16T03:27:00Z)

USB conection for SD ends with Emergency Mode after unplugging USB connector

regards

### Expected Behavior

Screen not showing "USB" - only white background

### Steps To Reproduce

1. power on radio
2. choose radiosetup USB Mode (ask or) Storage
3. plug in USB cable into computer
4. unplug USB cable

### Version

2.7.1

### Transmitter

Radiomaster TX16S

pfeerick, 2022-06-16:
This sounds like it somewhat relates to #2049, and may also be fixed by #2048?

raphaelcoeffic, 2022-06-16:
@pfeerick I believe it is, at least this is what my tests are saying.

closed 2022-06-17 by raphaelcoeffic