EdgeTX/edgetx #2062: Option for touch friendly graphics

factoryroll, 2022-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The new touch-friendly graphics for EdgeTX 2.8 look great for touch users, but displays a lot less information on screen. 

### Describe the solution you'd like

I'd like to see an option to switch between the new 'touch friendly interface' for users who prefer using the touch screen, and the more traditional OpenTX/EdgeTX 2.7 and less layout which fits more information into menus, lists and output views. 

### Additional context

Having this feature as an option would allow greater accessibility for those who cannot easily navigate the touch screen, or those who prefer to use the radio buttons to navigate.

raphaelcoeffic, 2022-06-17:
I know this is quite a change. We’ve put something like 6 months into it to exchange the underlying graphics lib and make it better. The truth is we cannot have it both ways per option. The complete code is now different.

so the proper question for me would be: what is actually missing? It works just the same with buttons. In fact, some aspects work with buttons now much better than they used to (ex: visual keyboard).

factoryroll, 2022-06-18:
Thanks Raphael, in my opinion what is missing in the 2.8 changes is the ability to look at the screen and see the maximum amount of multiple lines of the mixes, inputs, outputs. If it were possible, this might be achieved with some kind of compact view until the screen is touched? Maybe a radio without the colour touch screen running Edge would be better for me.  If it can't be both ways then so be it! I'll stick with 2.7 until the temptation of new features is too much. Appreciate your work and the response.

offer-shmuely, 2022-07-10:
I also miss the compact data presentation 
On one model I even can not see all the lines that build a single mixer channel, I need to go up and down many times to view info.

graph page can show 1.5 lines, so looking for the correct graph involve in a lot of up and down.

compact display will be a blessing

brainbubblersbest, 2022-09-23:
The Navigation through the Menu via rotary encoder was fast as fuck on the Old opentx Gui.
Now the Cursor is jumping from top to down. Then somewhere between from side to side. And again from top to down. This is a bit Irritating here and there. But really not a  issue. I had the expectation that could be changed over a different Design. Maybe the Human Brian just need some Training to get familiar with the New Gui.

pfeerick, 2022-09-23:
The cursor does need to be tamed down a little bit, but it is better than it was when there was no rotary acceleration. 😆 We do intend to try and make things a little more compact, but have no plans to support multi UI variations. One is enough. And, if we continue to flesh out the Lua API enough, there is no need ;)

offer-shmuely, 2022-09-23:
Is it possible on next version (i will probably skip this one) to add to a parameter that will control only the extra margin/spacing? Not totally two different versions of gui.
It may even make sense to upgrade the capabilities of THEMES to control that spacing, and not just colors

StefanKellerAC, 2022-10-26:
The title of this issue is misleading for me, but the discussion hits my concern.
Touch friendly is fine, but in doubt I would prefer a better overview, (or a choice :-)

In OTX (on Horus X10) there is one screen where you can see all Flightmodes 
with their settings for trims and fading.
You can easily spot similarities and differences:
[image]

In ETX 2.7 you had to scroll to see all flightmodes:
(and it was not shown which trim you are editing, but that would be easy to fix)
[image]

Now in ETX 2.8 you have to go through 3 screens to see and edit the trim settings of one single flightmode:
[image]
[image]
[image]

What is the Purpose of the button **"Check FM Trims"**? In simulator I can't see an effect.

malinhiles, 2022-10-26:
Glad to see some other activity around this, i'm also concerned about the touch interface UI update.

@StefanKellerAC in the above comment has a pretty good example of what I am not looking forward to. I'm not sure that i'm convinced that delving into multiple submenus with limited parameter visibility and accessibility is a benefit to user experience considering how this OS is used. I feel like that's the case regardless of whether you are using rotary encoder or touchscreen.

And sure, the rotary encoder and buttons still work as they did before, but you'd be remiss to use the buttons and scroll wheel in a new UI interface like 2.8. It's just so inconvenient and cumbersome.

I bought my radiomaster tx16s specifically for its modern, well thought-out form, combined with a solid implementation of the scroll interface in OTX. After having overall negative UI experiences with both touch-only radios (14SGA, 16SZ, and 32MZ) and button-only radios (OG X9d) in the past, not being able to stick with the current compact and dense UI in future updates makes me wary that the scroll based radios (including mine) will be made pretty inconvenient unless we live in the past with 2.7.

gagarinlg, 2022-10-26:
Feel free to propose another flight mode screen design that is also touch friendly. You could also implement it, if you have the skills.
There will be no more major changes to 2.8 though

malinhiles, 2022-10-30:
> Feel free to propose another flight mode screen design that is also touch friendly. You could also implement it, if you have the skills. There will be no more major changes to 2.8 though

I don't have the skill or knowledge base to rework any part of this, but I do spend hours a day using my radio for work, so I can see how UI changes like this could impact me and others from a mile away. Here's a couple ideas building off of @StefanKellerAC 's flight mode screen examples and photos:

[image]
This isn't far from being great for touch and scroll users. Instead of 8 flight modes on screen at a time like in this OTX picture, space them out vertically so that only 4-5 fit on the screen at the same time. You'll still see all of the relevant info for each flight mode, scroll wheel users can use the radio like they always have, and touch users now have more space to tap on a flight mode and delve into a touch-based edit interface if they want. And with some of the 2.8 rounded corners it'd be new, bitchin, and blue.

[image]
This picture also isn't far from being great for both touch and scroll users. Scroll based users can use it like they always have, and if you want to touch, just make it so that if you touch anywhere in a flight mode's "light-blue rectangle", it can take the touch user into a touch-based edit interface like what you've currently developed. Similar to the other idea, this idea doesn't interfere with scroll functionality, and allows touch users to benefit from the new touch interface you made. In comparing this idea to the current 2.8 implementation, in 2.8, the touch area to select a flight mode gets vastly smaller for touch users, all of the info at a glance is gone so its difficult to see what you've set up without jumping through nested windows, and scrolling becomes notably more inconvenient.  

It would be nice if you could see more than 2 flight modes at a time for this example, but I can see and am willing to accept that there is compromise made here for clarity so that all of the parameters are labeled. I like that. 

Both of these ideas aim to make it so that scroll users aren't going to struggle in a strongly touch-focused environment, and touch users aren't going to struggle in a strongly scroll-focused environment.

pfeerick, 2022-10-31:
Another possibility - revert to something somewhat similar to OTX flight mode screen, and instead of trying to have both edit and overview on the one screen, have them separate as they are now. In other words, it can be like the outputs screen - you can see multiple lines of information, but have to into the edit screen to make changes. 

One thing to keep in mind here is there are radios we support already (NV14/EL18), and radios coming soon, which are touch *only* - so making the UI usable when you can't scroll/don't have hard keys is going to have some priority (as well as trying to make it look nice and not so much like everything including the kitchen sink is being crammed in). You can have non-touch-friendly UI compactness when you can rely on alternative input like the rotary encoder or hard keys being present. The juggling act is going to make the right compromises for both.

offer-shmuely, 2022-11-01:
That is a very interesting idea.
To make the first screen on each tab, an overview, that is NOT touch friendly, and then if needed, enter an edit screen that will be a touch friendly

StefanKellerAC, 2022-11-01:
Proposal: if you touch somewhere on that overview screen, a touch friendly “zoomed” view around this point opens. 
With swiping, you can move that area if you missed the wanted item.
Are pinch gestures possible?

raphaelcoeffic, 2022-11-01:
> Proposal: if you touch somewhere on that overview screen, a touch friendly “zoomed” view around this point opens. With swiping, you can move that area if you missed the wanted item. Are pinch gestures possible?

Just in case that wasn't clear for everyone: your iPhone uses a processor that is **a lot** faster than what we have here. And we are already pretty close to what is doable with the hardware. So please keep this in mind when talking about this kind of things.

Also, adding extra clicks to do anything is probably not the best idea from the UX point of view. Try to check how your smart phone work to get some idea of how this gets done. We should not consider touch-less device as being something to strive for. These devices are dying (can you buy touch-less blackberry with the keyboard? was it convenient to use?).

raphaelcoeffic, 2022-11-01:
> it can be like the outputs screen - you can see multiple lines of information, but have to into the edit screen to make changes.

Having the *"flight mode"* screen similar to "outputs" would be easily doable, and would make things more consistent. We just need to decide which info should be displayed, at best only the most important, to avoid surcharging the screen.

offer-shmuely, 2022-11-01:
since we need to scroll the overview screen (it can show only 9 lines)
I think that some kind of  “edit” button will be needed to in-order move from overview mode, to the “touch friendly edit mode”

offer-shmuely, 2022-11-02:
> We just need to decide which info should be displayed, at best only the most important, to avoid surcharging the screen.

I believe that this operating system should not left the super users behind (like the v2.8 do), 
we are not here to create a simple email reader for everyone with minimal functions, 
we like to keep the os for complicated model as well, and we need for that and overview (on the flight mode/inputs/mixer/graph) with as many info as we can get.
 the user that do not need this info, will ignore it anyhow.

raphaelcoeffic, 2022-11-02:
> simple email reader

Most email readers cook coffee these days ;-) And many RC users tell me all this is much too complex, they "just wanna fly", no craft computer programs.

Obviously, we need a balance.

> complicated model as well, and we need for that and overview (on the flight mode/inputs/mixer/graph) with as many info as we can get

Let's ask candidly: what is the overview used for?
- a memory help on how the model works?
- trying to understand why something behaves the way it does?
- checking if everything is still properly configured?
- offering a choice of selections before editing something?

What are the bits of information missing on these overviews? (for each screen please)

Side question: are you ready to sacrifice speed/usability for more info on screen?

phileville, 2022-11-17:
I agree with the proposal to replace with initial flight mode screen with the uneditable FM overview screen similar to what is shown in "idea1" above, then when you select a particular flight mode from that screen, it goes to the current flight mode settings screen as it is in ETX v2.8.  That way we have best of both worlds and easy overview and touch friendly configuration page.

offer-shmuely, 2023-01-06:
nice idea, but it is still only 4 of them, can get at least 6?

offer-shmuely, 2023-02-10:
> Let's ask candidly: what is the overview used for?
>
> * a memory help on how the model works?
> * trying to understand why something behaves the way it does?
> * checking if everything is still properly [...]

Hi
you greatly described it, for me, yes, the first 3
* a memory help on how the model works!! (the mixsing&input are complicated!)
* trying to understand why something behaves the way it does (e.g. why specific FM override another)
* checking if everything is still properly configured.

> Side question: are you ready to sacrifice speed/usability for more info on screen?
yep, if that a must, yes

offer-shmuely, 2023-03-09:
any chance to progress with the overview/more-compress look & feel?

closed 2023-03-27 by pfeerick

philmoz, 2023-03-27:
> is it possible to make the title part of the page that scroll up

Not easily with the current system.
It is part of the header not the page content, so changing it would affect every model and radio settings page.

pfeerick, 2023-03-28:
Given the UI changes planned for 3.0, I don't think any further major work on these views is warranted at the present time.

offer-shmuely, 2023-03-28:
:-(

PR #3344 (2023-03-27): feat(color) - Change Flight Mode list and edit views.
Fixes #2062 

Summary of changes:
- Move trim setup to FM edit page instead of being a popup
- Add trim value display/edit box to FM edit page (only displayed if enabled)
- Display FM list like logical switches, show all FM data for each mode
- Update the trim value range limits in Companion to match the radio

The list view layout is a bit messy; but I'm not sure how to do it any cleaner.
The list view can only handle 6 trim switches at the moment - I hope there aren't any radios with more.

The trim values in the list view will update aut [... cut]