EdgeTX/edgetx #2063: Horus X12S Touchscreen. Wrong function selected.

KevinFairgrieve, 2022-06-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When selecting a menu from the eight main images. The wrong one is selected.
The screen is the locked from touch.
Going in to a menu using the rotary encoder I.E. MDL and then selecting the telemetry icon. RF set  up is displayed, again touch is disabled. 

Please see the video it makes it much clearer.
[https://youtu.be/1cRtON4a6s8](url)

### Expected Behavior

When the relevant item is selected it should go to that screen.

### Steps To Reproduce

See Video

### Version

Other (Please specify below)

### Transmitter

FrSky X12

### Anything else?

Horus X12S with touchscreen mod.
FW: edge tx-x12s
VERS: 2.8.0-nightly (c4fc11b7)
Date: 2022-06-18
Time: 03:17:33
CFG: EPROM_VER

raphaelcoeffic, 2022-06-19:
That pretty much sounds like you mounted the touch sensor upside down.

raphaelcoeffic, 2022-06-19:
Or the touch sensor outputs the coordinates inverted.

KevinFairgrieve, 2022-06-19:
Using the firmware from "https://github.com/wimalopaan/Electronics/tree/main/edgetx/firmware/x12s" I have had no issues, as per this video.

[https://www.youtube.com/watch?v=hiQ0qBhCax8](url)

wimalopaan, 2022-06-19:
Just tested witch actual `main` and it works as expected

KevinFairgrieve, 2022-06-19:
main?

Can you add an EN version so that I can try that?

Many thanks.

KevinFairgrieve, 2022-06-19:
> Just tested witch actual `main` and it works as expected

main?

Can you add an EN version so that I can try that?

Many thanks.

rotorman, 2022-06-19:
main means main branch of EdgeTX. For building on your own, please see: https://github.com/EdgeTX/edgetx/wiki/Building-radio-firmware-in-a-webbrowser-with-Gitpod

wimalopaan, 2022-06-19:
Just use my german version to test, if you can select the functions

KevinFairgrieve, 2022-06-19:
> […](#)

Will try tomorrow.

Thanks again.

pfeerick, 2022-06-20:
> main?
> 
> Can you add an EN version so that I can try that?

i.e. For a pre-build main you can choose the latest build from [here](https://github.com/EdgeTX/edgetx/actions/workflows/actions.yml?query=branch%3Amain), or use a [nightly build](https://github.com/EdgeTX/edgetx/releases/tag/nightly) or grab

KevinFairgrieve, 2022-06-20:
> > […](#)
> 
> Will try tomorrow.
> 
> Thanks again.

Tried the DE version, still the same.

wimalopaan, 2022-06-20:
Then something of your hw is screwd up

KevinFairgrieve, 2022-06-20:
And I have now tried the latest "main" [3ce1aef](https://github.com/EdgeTX/edgetx/commit/3ce1aefc447476366aa7cd361b28638d95d792fd)

Still the same.

Pretty sure my screen is in the correct way, but a stripdown will show that.

pfeerick, 2022-06-20:
Check the touch screen diagnostics first ...  Radio Settings -> Hardware -> Analogs (at the very bottom). If you press and hold in the middle of the screen, and then move up to the top left corner, coordinates should be near 0,0, and near 480,272 at the bottom right corner

KevinFairgrieve, 2022-06-21:
> Check the touch screen diagnostics first ... Radio Settings -> Hardware -> Analogs (at the very bottom). If you press and hold in the middle of the screen, and then move up to the top left corner, coordinates should be near 0,0, and near 480,272 at the bottom right corner

So that appears to be my issue. If I move to top left the cross goes to bottom right. How to fix?

Have I wired something wrong?

raphaelcoeffic, 2022-06-21:
> So that appears to be my issue. If I move to top left the cross goes to bottom right. How to fix?
> 
> Have I wired something wrong?

possibly, you should check how the sensor is oriented. Other possibility would be that it’s configuration stored inside is wrong.

rcman45, 2022-06-23:
si le capteur est mal monté ou si sa configuration interne est incorrecte alors pourquoi fonctionne-t’il avec la version de wimalopaan

pfeerick, 2022-06-23:
Given you mention that `etx_x12s_de_2.8.0_wm_touch_4.bin` and `etx_x12s_de_2.8.0_wm_touch_5.bin` didn't work for you, but `etx_x12s_en_2.8.0_wm_touch_3.bin` did, it could be due to changes as the PR for this evolved - hence why internal configuration of the TP itself could be wrong / at fault. I think the early version of wimalopaan's PR may have set the rotation bit every time, rather than just when it was different to what was expected, or had some other rotation flag enabled... which may not have have been detected while it was being tested since it probably wasn't done with an as new screen (until now).

closed 2022-09-19 by pfeerick

KevinFairgrieve, 2022-09-23:
I just saw that an uninverted version was available.
https://github.com/wimalopaan/Electronics/tree/main/edgetx/firmware/x12s

This has now solved my issue. I hope this will be rolled out to the full release. Currently only available as a DE version.

pfeerick, 2022-09-24:
Rather than just hope, why not ask @wimalopaan *what was configured differently in that firmware build?* ;)

raphaelcoeffic, 2022-09-24:
I believe the difference is really in the config stored on the touch controller, depending on which versions have been installed in between. My plan was to to see if I could download the existing config to compare and upload a new version if required.

KevinFairgrieve, 2022-09-24:
> Rather than just hope, why not ask @wimalopaan _what was configured differently in that firmware build?_ ;)

I have now done so. Thank you.

pfeerick, 2022-09-25:
@KevinFairgrieve Are you in a position where you can see if https://github.com/EdgeTX/edgetx/pull/2351 will fix this for you? Basically, if that the TP input is still inverted with the build of firmware from that PR, there is now a mechanism by which you can reset the touch panel controller to find out if it is the problem. You should be able to plug the radio into your computer, choose the `USB Serial (VCP)` option, and then use a serial console tool like Putty to connect to the virtual serial port that is created (at 400,000 baud). When you press enter after opening the console up, you should get a prompt. You enter the `reset_gt911` command, and press enter. It should tell you what the GT911 configuration value was, and what it is now. Hopefully, the touchpanel is also now the right way around. 

If that seems too daunting, that's ok. And if you do want to give it a go, I'll try and walk you through it more if needed.

KevinFairgrieve, 2022-09-25:
> @KevinFairgrieve Are you in a position where you can see if #2351 / tomorrows nightly build will fix this for you? Basically, if that the TP input is still inverted with the build of firmware from that PR, there is now a mechanism by which you can reset the touch panel controller to find out if it is the problem. You should be able to plug the radio into your computer, choose the `USB Serial (VCP)` option, and then use a serial console tool like Putty to connect to the virtual serial port that is created (at 400,000 baud). When you press enter after opening the console up, you should get a prompt. You enter the `reset_gt911` command, and press enter. It should tell you what the GT911 configuration value was, and what it is now. Hopefully, the touchpanel is also now the right way around.
> 
> If that seems too daunting, that's ok. And if you do want to give it a go, I'll try and walk you through it more if needed.

Happy to give it a go.

I will download "putty" and the next nightly.

KevinFairgrieve, 2022-09-26:
Downloaded the Nightly 83c21ad.

Loaded and all works as expected.

I now see the 2.8.0 RC1 is also out.

Luznatural, 2023-02-09:
I have exactly the same problem with my X12S touch screen. Everything is inverted.
I have installed PuTTY but I am unable to make the virtual serial connection with the radio, as apparently Windows does not detect it.
So I cannot reset the screen.
Is there another way to reset the screen? Perhaps making some lua process to reset the screen without having to get a connection with a computer?
I would appreciate some help here
Thanks in advance

rotorman, 2023-02-09:
You could try hTerm https://www.der-hammer.info/pages/terminal.html instead (this is a serial program that I mostly use).

Luznatural, 2023-02-09:
I have been able to load the drivers and connect with COM4 to the radio WITH 400.000 bauds, but I am unable to write anything in the console of PuTTY. When I type, nothing happens. Any ideas?

raphaelcoeffic, 2023-02-09:
@Luznatural most probably CLI is not set for the USB serial port. Go to the radio hardware settings, at the end of the page.

Luznatural, 2023-02-09:
I have in the hardware tab of SYS in the X12S the USB-VCP to CLI, Aux1 OFF and port power off, and no luck.
Can you be so kind to Ispecify for me the connection data in PuTTY?
I have the following:
Serial line COM4
Speed (baud): 400000
Data bits:8
Stop bits: 1
Parity: None
Row control: XON/XOFF

More ideas?
Thanks in advance

rotorman, 2023-02-10:
Row control? You mean flow?
Anyhow, please try with hTerm instead, as with this all the settings are cleary visible on one page and if still no luck you could post the screenshot and one can easily see what the issue would be.

pfeerick, 2023-02-10:
I use Kitty, which is just a fork of Putty, and those settings look correct.

[image]

When you start the session, hit enter once, you should get the `>` prompt

[image]

I can't remember the default terminal settings on kitty/putty, you may want to check that also  - specifically the Implicity CR/LF settings

[image]

Luznatural, 2023-02-11:
First, thanks to all for offering your savior counseils.
I have tried PuTTY, KiTTY and hTerm without success in my 3 different computers. It looks like my radio has something wrong with the VCP connection or something extrange (like myself) makes me unable to reset the tuch screen.
I have also tried several different connection cables, just checking if it was a wiring problem.
No luck up to now.
I would like to know if you think there could be any alternate way to reset the screen using any command or making something that could modify it inside ETX, perhaps a LUA script, or producing a specific version of ETX for X12S that includes this reset in the core of the firmare.
Please, help this poor and desperate soul
Thanks in advance

KevinFairgrieve, 2023-02-11:
Just as an off chance, have you tried using one of the earlier versions from wimalopaan?

I found that using the test versions was OK then when 2.8.0 came out, mine just started working ok.

See my post above.

https://github.com/EdgeTX/edgetx/issues/2063#issuecomment-1159625809

It may be worth a shot.

rotorman, 2023-02-11:
@Luznatural can you show a screenshot of your hTerm?
Especially, did you set CR-LF for `Send on Enter` in the transmit side:
[image]

Luznatural, 2023-02-11:
> @Luznatural can you show a screenshot of your hTerm? Especially, did you set CR-LF for `Send on Enter` in the transmit side: [image]

Here you have a screenshot:
[image]
No luck connecting with the radio

Luznatural, 2023-02-11:
> Just as an off chance, have you tried using one of the earlier versions from wimalopaan?
> 
> I found that using the test versions was OK then when 2.8.0 came out, mine just started working ok.
> 
> See my post above.
> 
> [#2063 (comment)](https://github.com/EdgeTX/edgetx/issues/2063#issuecomment-1159625809)
> 
> It may be worth a shot.

I have not been able to find any file from @wimalopaan, except one named etx_x12s_de_2.8.0_wm_7_tuninv.bin that I tried also without luck. Everything continues the same.
The screen continues inverted.

I have used Windows 11 and Windows 8 to run PuTTY, KiTTY aand H-term. Could that be part of the problem?

Thanks to all for your help.

rotorman, 2023-02-11:
@Luznatural Your baudrate looks wrong. Disconnect, change it to 400000, reconnect. Then put your cursor to the right of `Type "ASC"` field and press Enter there. Any improvement?

Luznatural, 2023-02-11:
Yes!!!
Finaly I was able to connect and pressed reset_gt911 and the radio responded something like reset_gt911, your current configuration is "0" and then the screen started to work correctly.
Thanks to all for your patience with this very dumb guy.
As an aliby, I didn´t even remember how long ago I used a COM terminal.
Thanks again for your kind help

KevinFairgrieve, 2023-02-12:
Excellent. Now your TX12S will be the transmitter it should have been from the start.

If only it ran ETHOS as well!

mervin1961, 2023-06-12:
Hi,
I just bought the gt911 touch panel for my Jumper T16 and I had the same problem. Reading this post I tried to reset gt911. It return with old version 109 and new version is 0 it turn the right side but from the center 2 to 2.5 cm the screen is inactive. Is there something else I can do to correct this? Thanks.

rotorman, 2023-06-12:
You can diagnose touch screen under hardware analogs diagnose page. Is the 0,0 point somewhere in the middle or is an area on one side cut? Would be easier to understand if you could make a small video tapping the screen at various points, while the analog diagnose page is open and post a link to that video here to see if it is a software or hardware issue.

mervin1961, 2023-06-13:
Firstly thanks for the reply. 
Check with diagnostics the left top in giving the maximum 81:0 that is if my finger is extreme top and extreme left and the right bottom is 479:271. Centre nothing happens. I have Hterm to reset the touch panel and the is the only software that works for me.  I have already made a video and I will upload it here. Sorry for the video quality. I had to compress it from a 88mb to upload it here.

https://github.com/EdgeTX/edgetx/assets/92143050/6e18c896-d1e2-414e-ab64-e1be8ae5594c

mervin1961, 2023-06-13:
As you can see the main menu screen the 4 center selections icons are dead. As you already know before I did a reset the gt911 was diagonally opposite on the selection.

rotorman, 2023-06-13:
Can you please navigate first to analogs diagnose page and THEN tap the screen while doing a new video. You should see on the bottom of the screen the detected coordinates of your finger. I am interested to see which coordinates get detected and which not.

mervin1961, 2023-06-13:
Uploading now. Please let me know if you need more videos. Thank you.

https://github.com/EdgeTX/edgetx/assets/92143050/3070d49a-73da-499f-895c-4461d4999eae

rotorman, 2023-06-13:
Can you try once more resetting the GT911 via CLI.
If that does not solve the issue, it seems to me that your touch panel layer is unfortunately defective.

mervin1961, 2023-06-13:
Everything works before resetting only diagonally opposite or 180 degrees around.
Actually I have tried resetting many times but I will try again in fact hoping everything will be ok. Give me a few mins I will do another video.

rotorman, 2023-06-13:
The video is good now, thanks. Shows what I was after. What EdgeTX version do you have installed - how does the VERSION page look like?

mervin1961, 2023-06-13:
[image]

rotorman, 2023-06-13:
@Stinky80 does it work for you with T16 and latest dev branch (I do not have T16 myself to test unfortunately)?

Stinky80, 2023-06-13:
in stabile reliases edgetx.  t16pro working
--
Отправлено из Mail.ru для Android вторник,

mervin1961, 2023-06-13:
Yes it seem to be working alright. So far haven't found any faults. 
All you guys in here are doing wonderful things.
Thank you.

mervin1961, 2023-06-13:
Anyway I am chatting with the seller on aliexpress regarding this issue. Hope this can be resolve. I have 2 units of this lcd one original in the Jumper and the other I bought maybe 3 or 4 years ago and I tried the display is correct installed ribbon upwards in the T16. Looks like radiomaster have some kind of agreement with the lcd suppliers because I can't find the lcd's anymore except for bulk order from abibaba. All the lcd's are 180 degrees around meaning correct view with ribbon installed downwards. I buy this because it is cheap for me. I live in Malaysia and retired and our currency is pretty bad now. Original touch panel from radiomasterrc.com would cost me about RM130 with shipping compared to gt911 only RM50.
Anyway thanks for you help. Really appreciated it.

Luznatural, 2025-02-12:
Just for general information:

The easiest way to reset the inverted screen for X12S radios is using any Android tablet/phone.

In my case, I used a very simple and free application named "Serial USB Terminal" by Kai Morich, that you can download free from Play Store. I do not know if there is something similar for Apple phones/tablets.

Once Installed, you simply open the program, connect the radio to the tablet/phone, and select in the radio the opción VCP (serial).
As the program takes care automatically of the parameters of the connection, you do not need to worry about baud rates, COM ports, CR+LF, etc.

You simply touch the intro line, and the connection is established. Then write "reset_GT911" in the most lower line, press the arrow (intro) and the screen shows the "reset to 0" response, and then the screen works perfectly.
Disconnect everything and you are ready to go.

Using Windows for doing that is becoming more and more complicated, as very frequently Windows does not allow easy access to COM ports, as those ports are now being used for some other programs, and very frequently you do not get access to create correctly the COM port for the radio.