EdgeTX/edgetx #2065: Enhance wiki: Flysky Hall gimbals FS-HZCZ03-ADJ on Jumper T16

stefan73, 2022-06-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The current wiki on using the Flysky CNC Hall gimbals is very incomplete when it comes to modding the Jumper T16. ( https://github.com/EdgeTX/edgetx/wiki/Flysky-Hall-Sticks-Mod )

Issues:
Mechanical complexities not mentioned.
Deltas in electrical wiring as Radiomaster and Jumper use different gimbal connectors / pinouts.

### Describe the solution you'd like

I have written a technical note on how to mod the Jumper T16; linked below. Feel free to add this to the edgetx repository, steal from it to enhance the wiki (or whatever is needed to help).

https://drive.google.com/file/d/1pup3YBhwhN8K9T0jPQi099Klm51RdCGr/view?usp=sharing

### Describe alternatives you've considered

n.a.

rotorman, 2022-06-19:
@stefan73 awesome work! Would you be able to provide a markdown version of the text, ideally embedded into https://github.com/EdgeTX/edgetx/wiki/Flysky-Hall-Sticks-Mod ?
You can git clone the Wiki with:
`git clone https://github.com/EdgeTX/edgetx.wiki.git edgetx_wiki`
If you send me the new file or a diff, I can update the EdgeTX Dev Wiki accordingly.

stefan73, 2022-06-20:
Never used markdown. Should be possible, might though take some time.