EdgeTX/edgetx #2066: Jumper T18 Pro EdgeTX 2.7.1 No SD card Error

adamkerrnz, 2022-06-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Tested radio using stock T18 FW (From Jumper website) and SD card is working.

Flashed 2.7.1 following text guide exactly:
Started with a fresh Fat32 Formatted SD Card (Which was also working with stock FW)
1. Installed Horus SD Card contents
2. Flashed EdgeTX Bootloader
3. Flashed Firmware from EdgeTX Bootloader

When powering on radio, there is a No SD Card error.

Note this same behaviour was shown when attempting to install OpenTX FW (Not supplied by Jumper)

### Expected Behavior

There does not appear to be any issue with HW or the SD card, so assuming this is to do with either a bootloader or FW issue.

### Steps To Reproduce

Started with a fresh Fat32 Formatted SD Card (Which was also working with stock FW)
1. Installed Horus SD Card contents
2. Flashed EdgeTX Bootloader
3. Flashed Firmware from EdgeTX Bootloader

### Version

2.7.1

### Transmitter

Jumper T18

Eldenroot, 2022-06-19:
I can confirm this on X9D+ 2019 SE

All was fine but suddenly No sd card error and now I cannot use the transmitter. Is there any way how to fix it?

pfeerick, 2022-08-19:
@adamkerrnz Can you try again with the next nightly build to see if this is resolved for you?

anshchawla521, 2023-02-23:
I am facing the same issue.
transmitter: Radioking tx18

any fixes for this ?

Eldenroot, 2023-02-23:
Any way how to fix? Taranis cannot even reflash when this error is displayed. Any hints?

raphaelcoeffic, 2023-02-23:
When this is displayed, that mean that the SD card could not be read. Most of the time, this is due to something physical (slot / contacts). Sometimes the card is badly formatted. I don’t think there is anything that can be / should be fixed in software here.

raphaelcoeffic, 2023-02-23:
And for those with Radioking TX18: sorry guys, this radio has serious hardware issues and we cannot possibly support it, because of the erratic hardware.

anshchawla521, 2023-02-23:
In my case the Sd card is plugged in and it still says no Sd card.

any guesses on which part of the hardware I should look into ?

anshchawla521, 2023-02-23:
When I reflash open Tx it works flawlessly.

anshchawla521, 2023-02-23:
I am using open tx 2.3.8 which was used for the jumper t18 radio

raphaelcoeffic, 2023-02-23:
FYI: most issues with SD card are due to:
- fake SD chips
- Bad contact (cheap slot)
- Formatting issues

anshchawla521, 2023-02-23:
> OpenTx and EdgeTX use the same code at this place. So it cannot really be anything software related. Please contact the non-existing support of the disappeared vendor if you need support for this product, or just use whatever works for you.

😂😂😓😓

richardclli, 2023-02-24:
Open the back cover and check the SD card slot component, if the soldering points is not visible outside the slot (top of the slot), you have a failure SD card slot, this version has a problem that the SD card detection is intermittent.  The OpenTX firmware works because it is from the manufacturer and they modify the code to cancel the detection of the SD card insertion.

richardclli, 2023-02-24:
You can check if your card slot is just like this one.  And you can find this photo provides a hardware mod that hide the problem.  But this mod CANNOT GUARANTEE FROM DATA LOST due to failure hardware.

[image]

anshchawla521, 2023-02-24:
> You can check if your card slot is just like this one.  And you can find this photo provides a hardware mod that hide the problem.  But this mod CANNOT GUARANTEE FROM DATA LOST due to failure hardware.  The photo show the failure card slot part, so you may compare with your own radio, if it is the same, then 100% guaranteed you have this issue.
> 
> [image]
> 

[image]
I would say the soldering looks fine so that shouldn't be an issue

raphaelcoeffic, 2023-02-24:
And still you Radio only works with the "special" workaround... So I'd say this exactly the issue you're having.

What is 100% clear: we won't remove the card detection code just for your borked radio to work "somehow".

anshchawla521, 2023-02-24:
> And still you Radio only works with the "special" workaround... So I'd say this is exactly the issue you're having.
> 
> What is 100% clear: we won't remove the card detection code just for your borked radio to work "somehow".

Thank you bro.
Your mod worked and what basically it does is forces the sdcard_detect pin low thus telling edge tx that SD card is present even when it is not.

richardclli, 2023-02-24:
What your photo shows is the new version of card slot.  It should be more reliable than the one show in my photo.  Maybe you can try to change a better SD card as well.  I have some reports that the SD card comes with the radio is not reliable as well.

pfeerick, 2023-03-17:
Closing this issue as everything is pointing to this being a hardware issue. Feel free to continue any discussion regardless.

closed 2023-03-17 by pfeerick

JINODK, 2023-03-17:
Just to provide some infomation on this topic, I also got this error on my Taranis X9D+ 2019, but for a very different reason,
Having gone through the trouble of recompiling the code myself to trick the detection function into thinking that the SD card was inserted at all time by overriding `#define SD_CARD_PRESENT() true` in the corresponding *`board.h`* header file, that to no avail
I realized that although my SD card was correctly formatted as **`FAT32`**, ***the SD card needs to be in MBR and not GPT***, after changing it to MBR, it works just fine with the default build of the firmware.

pfeerick, 2023-03-19:
Ouch! Nasty one indeed! 😮

Grandbill59, 2025-05-31:
**I have sdcard issue on Jumper T18 too** , only when I want to use Edgtex.
I bought a jumper T18 pro, running opentx t18 2.3.8, which works well.
I tried to jump to edgetx 2.11.1 and up to date FW (last release).
I used the sdcard method to upgrade.
### 
1) using the original 1Go sdcard, it works a few minutes bind with receiver SxR ok, setting screens ok but  then I got crash using SxR lua, then it works , then it crashes...
restarting the T18, I got no sdcard at the beginning, or after a few minutes using the GUI.
2) I format the sdcard reload it  -> same pb , no sdcard at start, or rarely after a few seconds 
3) try with 
      Kingstone 100Mb/s 32Gb -> same pb , no sdcard at start
                 trying with a tape on the card to thick it a little bit  -> same pb , no sdcard at start
      SanDIsk ultra 140Mb/s 64 Gb -> same pb , no sdcard at start
      SanDisk ultra 120 Mb/s 32Gb -> same pb , no sdcard at start
4) comming back to the original sd 1gB, loaded with opentx (backup the original sd)
   ====> it works without any pb since more than 1 hour.

**I think there is a real issue with sd card reader on edgtx.**