EdgeTX/edgetx #2069: Add option to erease EEPROM data

gagarinlg, 2022-06-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When a radios that has been set up with a pre-yaml firmware it is impossible to do a reset to defaults, because the EEPROM data will be converted and used again.

### Describe the solution you'd like

Add an option, either in the boot loader or somewhere else (radio settings) to allow the users to erease the EEPROM.

pfeerick, 2022-06-20:
That would also be a good time to handle re-adding https://github.com/EdgeTX/edgetx/issues/1521

gagarinlg, 2023-03-18:
closed, because 2.9 does not support binary conversion anymore

closed 2023-03-18 by gagarinlg