EdgeTX/edgetx #2071: Play Value does not include overrides

mh125, 2022-06-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The special function Play Value ("Play Val") does not include overrides.

I have two functions that override CH3 ("Overr. CH3"), which is my throttle. However, when executing the special function "Play Val CH3", I hear the position of the stick, not the override. This is irrespective if the override is before or after the Play Value command in the special functions list.

### Expected Behavior

My expectation is that Play Value reads out the value that is actually sent out over the air.

### Steps To Reproduce

1. Create Special Function "SA^  Overr. CH3  99 on"
2. Create Special Function "SB^  Read Val CH3"
3. Go to a screen where you see the CH3 output that is sent (and that it is other than 99)
4. Activate SA and see that 99 is being sent
5. Activate SB and hear that a value other than 99 is read out

### Version

2.7.0

### Transmitter

Radiomaster Zorro

pfeerick, 2022-07-03:
Also TX16S, so is probably a systemic issue re: how override applies or how play value gets the value.

raphaelcoeffic, 2022-07-06:
This is indeed "by design".

Here is how it works:
- `playValue` uses the mixer channel values (`ex_chans[]` / `[-100;100]`)
- "override channel" sets the output (`channelOutputs[]`)

If you want the values to be read correctly, you should not use "override channel" (considered *evil*), but rather use a "replace" mixer line at the end of the mixer list for that channel with the proper switch instead.

mh125, 2022-07-06:
There are 3 main screens that show the output channels (incl. invert and override). That's correct, because I want to know what is actually being sent out. And instead of having a look at my transmitter, I want one of these values read out. How can you claim that wanting to know channelOutput orally is not good? (To be frank, his "won't fix, by design" sounds more like a "won't do, too lazy" to me.)

Why not increase the options for `playValue` so that is can read out mixer output `ex_chans[]` or channel outputs `channelOutputs[]`, depending on what the pilot wants to know?

pfeerick, 2022-07-06:
I actually agree here, and consider this to be a bug still, since if regardless of how the channel is being set (input, LS, override, trainer, etc), `playValue` is *NOT* playing the value of the channel output which is what you expect when you choose CH1, CH2, etc. and channel override is in effect. If you wanted the input value, you would have chosen that instead.

raphaelcoeffic, 2022-07-06:
@mh125 what I’m saying is: don’t use “Override Channel”. You can achieve the exact same thing with a mixer line doing the replacement by the using “Replace” instead of “Add”. Then everything will work as you expect.

raphaelcoeffic, 2022-07-06:
> things like channel inversion, offset, clipping also aren't taken into account for play value? So perhaps entries for outputs is what is needed?

that is right. “Play value” can only fetch and play mixer channels, not output.

raphaelcoeffic, 2022-07-06:
@mh125 to be more explicit, look at channel 3 on this screenshot:
[image]

This mixer line uses `REPLACE`, which means that the Mixer Channel value will be replaced if the switch is active, which is the same as the special function "Override Channel", except that it overrides the **Mixer Channel**, and not just the **Output**.
[image]

Normally, you deal with things in a somehow "logical" rather than physical way. For instance, when using the special function "Override Channel", settings in "Outputs" won't influence the value which is output to the servo as long as the special function is active. This is considered a flaw and is the reason why "Override Channel" was supposed to be phased out. Because if was popular and historically pre-dating the "Replace" mixers, it has been kept in the code.

pfeerick, 2022-07-06:
> historically pre-dating the "Replace" mixers, it has been kept in the code.

Ah, I didn't realise the timeline there, I had though it was the other way around... Looking back over some of the discussion... I can see both sides of the argument... and the dilemma the OTX were stuck with... as I can see the major benefit of override being it set the output to *exactly* what you set it to - so when you set X output, it will be X output, regardless of any inversion, limits, etc... it just is. Which is a problem, if it's not respected. 

So maybe we should start talking about if we should be depreciating it in a future version (i.e. 3.0) with automatic conversion to mixer replace lines on models that still use it, and see where that goes. :grin:

closed 2022-07-06 by pfeerick

mh125, 2022-07-06:
Thanks. I know how to replace in Mixes.

The discussion parted into two directions. Let me comment on both of them.

1. Override function
This function is not at all illogical. I use it for my emergency button. I know what values I want to send _exactly_. If my "send -100 for the emergency routine" would still go through the "Outputs" layer, the -100 could be changed by subtrim, min, max, direction, curve. I don't want any of that, as it is a source of potential error. I want to make sure that -100 is sent out – and minimize the chance of any configuration that accidentally prevents that.

2. Importance of knowing sent outputs
If you say that the sent output values are not important and "outputs cannot be used for anything as a source", then please explain why the main screens shows sent outputs (see lower half of these screens).
[image]
[image]
I regularly look at one of these output channels. Why shouldn't I be not allowed to get it read out to me?
With the arguments you put forward above, you state that nobody needs to know the output values. Well, I do. But then, if nobody (except me) needed it, the main screens wouldn't show the sent outputs, wouldn't they?

pfeerick, 2022-07-06:
re: point 2 - you have the wrong sauce there ;) Source as in source for a logical switch, input, widget, etc... not as in being able to view the value of an output. This is where I agree with you on play value needing to have access to the ouput values rather than just the mixer values.

mh125, 2022-07-06:
@raphaelcoeffic I think you did not get my point no 2. Its completely separate from what somebody thinks about "override channel". Even without the "override channel" function, I'd like "play value" to read out the actual output (inverted, clipped, shifted, "curved"), not the mixer values. I want my transmitter to read out the values that it already displays on the main pages.

Looking at this example, why do you think it is a bad idea if I want my transmitter to read out "12"? (This is an inverted channel.)
[image]

pfeerick, 2022-07-06:
Point 2 was the path of least of resistance as far as the SF was concerned, but the one I really didn't want to see. 

Instead of a separate function for point 1, is another option for `Play value` viable, or just as messy? i.e. it could have a standard source list (the default), and an extended/custom one.

raphaelcoeffic, 2022-07-07:
> Instead of a separate function for point 1, is another option for Play value viable, or just as messy? i.e. it could have a standard source list (the default), and an extended/custom one.

From I can see, this could be quite tricky and introduce a fair load of special casing in the code. Maybe some good idea will jump into my mind later on, but right now I don't see a good path to success.