EdgeTX/edgetx #2073: LVGL Mixer/Output Channel Numbers

jmxp69, 2022-06-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Using 2.8.0 with LVGL graphics on TX16s on the mixer and output screens when a label is applied, the channel # is no longer displayed on either screen.

I just got through setting up a model and found once I had a label assigned on an output channel, I had no way of correlating the output to the channel number on the mixer. Outputs allow duplicate label names. Likewise the channel number is removed from the mixer when an output label is applied. While I appreciate the simplicity of using labels vs numbers, there remains a dependence correlating the pin number on the RX to the channel number in the mixer and subsequently the output. Minimally, 

### Describe the solution you'd like

Minimally I believe it is a mistake to remove the channel number from the mixer when a label is applied to an output. Ideally, the channel number would remain on the mixer and outputs screen for ease of use. 

### Describe alternatives you've considered

Mixers and Outputs should both show the channel number. Setting up a model without this information was much more difficult than it should have been.

raphaelcoeffic, 2022-06-21:
> Minimally I believe it is a mistake to remove the channel number from the mixer when a label is applied to an output. Ideally, the channel number would remain on the mixer and outputs screen for ease of use.

While I understand your requirements, I'm starting to get issues buying some screen space ;-) In earlier versions, ppl complained they would see only the channel number and not the label. I'm trying to figure out how to have both ways, but this might get tough.

jmxp69, 2022-06-21:
I've setup a model from scratch in both scenarios. Without access to the actual channel number on the mixer it forces the user to revert to the receiver and remove/replace connectors to figure out what channel the mixer is utilizing. I think it will be a serious mistake to omit the channel number from the mixer when an output label is applied.

rburrow87, 2022-10-03:
Just giving this a poke to see if there have been any thoughts on a way to show the channel number and label at the same time. It's kind of a headache figuring out what's what during setup if you can only see one or the other when getting into something more complicated than a basic setup. Would be nice to see it rolled in before 2.8 releases.

I saw a suggestion of using `nn:name` when there's a label set in #2400

gagarinlg, 2022-10-03:
I am working on something:
[image]

Eldenroot, 2022-10-03:
@pfeerick - maybe should be added to 2.8 RC2 milestone? It looks like a regression.

gagarinlg, 2022-10-03:
See #2464 for screenshots, what I came up with, for now. I am open to suggestions to make it better

JimB40, 2022-10-04:
@pfeerick @gagarinlg  let's do not rush with that. I think there is little inconsistency across system with ID/Label
As you know I'm browsing through all UI options to make review, also checking possible UI space for data.

So base idea across system is that items have their have their IDs with prefix (identifier) so later with selection you can differnciate what you choose. Some of them have name (label) and once label is set this label is displayed 
interchangeably with identifier.

- Inputs (icon)
- Mixes, (CH) 
- Outputs (CH)
- Curves (CV)
- Global Variables (GV)
- Logical Switches (L)
- Special Functions (SF)
- Global Functions (GF)

But there is legacy mess with identifiers/naming/logic whatever you call it that creates confusion.

To give you example of hey "let's put another data on UI" :

These are clean model Mixer/Outputs
[image]
[image]
As you can see for first item both names are CH01

I've added to first Mixer item (channel) second mix line naming (labeling) both so I can differenciate them M01-1 & M01-2
[image]

You can see that mixer first item's name/label stays 'CH01'. Because it refers to channel identifier which has not logic connection in UI and it's used both in Mixers & Outputs Tab.
[image]

What's more It seems those Mixer lines names/labels are not used afterwards in any selection. So what they are for? 
[image]

Okay so how do I change not mentioned anywhere 'Channel' name/label. Well in Output Tab editing field 'name'. Let's give it label 'OUT01'
[image]
[image]

So now in Mixer first item label is 'OUT01' and this name instead of identifier (CH01) appears in selection
[image]
[image]

Okay let's test foolproofness

Output 1 & 2 are now labeled 'OUT01'
[image]
[image]
System is not checking label uniqueness so this can be created

To recap:
1. 'Name' is used for things that are later used for selectable and non-selectable items
2. Identyfing
 a. If 'Name' interchanges with identifier it should be displayed on selection list with identifier to make it clear 
  - Roll (CH01)
  - Arm (I02)
  - Flip (LS04) - if we ever introduce Logial Switches' names :)
 b names should be checked for uniqueness
3. There is concept of Channels used in Mixes & Outputs so at least it shoud be stated that Output editable 'name' field is in fact 'channel name' used then in Mixes Tab & Popup Selection list as a filter group
4. If something serves as tip/hint for user (like Input Lines name or Mixer Line name labels) - I mean it's not used then for selection - we should consider renaming to 'label' and display with label icon 

I'm seasoned user but it took me some time to solve what dev intention was.
Then no wonder new user can be confused.

Eldenroot, 2022-10-04:
Wau, thank you for your detailed input.

Yes, it is messy and confusing not only for a new users...

jmxp69, 2022-10-04:
Robert, This is a very clear diagram of the situation. I believe the only use of the Mix Name field pre-lvgl is to describe what thing should be moving for a given mix line where the channel number indicates what wire should be connected to what pin on the rx.

Interestingly, in pre-lvgl I can find no correlation between "mix name" in the mix editor and "name" in the output editor. From a functional point of view, I believe the linkages should be:

1) Most critical: Ch# on the left side of each mix group on the mixer page.
2) Mix name field in-tact for each mix line in the mixer. When programming mixes this is important as a visual reference to identify the purpose of a mix.
  50% I-AIL  PAIL
+50% I-ELE EVON
3) Ch# on each output. I don't see much value assigning a name to these lines because Ch1 can output either a single mix line (PAIL) or a combined mix line of one exists (EVON). So isn't necessarily a single mixer label apropos to an output. It's really the combined--lacking a label--mix being sent to the output.

gagarinlg, 2022-10-04:
I think the output naming is mostly good to remember which rudder/flap/... is connected to that output. If you have a mix of different stuff, which has inconsistent channel use, that is a big help, in my eyes.

Forcing the names to be unique is probably a bigger taks, because that uniqueness should than span over all mixer sources, includin switches, logical switches, ...

JimB40, 2022-10-04:
Seems clearly that both identifier and label are needed.
- Identifier is more technical (as you stated what wire to what pin)
- Label (name) is human friendly to give brief hint 'what was my idea'

In fact I use lines labels a lot like in this 'multi' input that allows me to Arm, Crash Flip, Engage AirMode automatically, Engage Launch mode & Switch on buzzer using one channel. So labels give me overview what is purpose of certain Logical Switch.
[image]

Thing is to name in UI correctly what is for selection & what just for info.

1. Agreed
2. Agreed - but clearly stated it's just information label
3. Agreed
a. haven't checked yet case when you select as source 'channel' - what value is actually passed. the one from mixer or the one from output. I guess latter one. For example in LUA reading CHxx you get MIX value not OUTPUT value.
b. Well here I'm stucked the most with base idea or Outputs. Maybe because I fly quads mostly and use OUTPUT only to correct CH1-4 end points. While I understand idea of INPUTS - OUTPUTS seems to have just one purpose - to make kind of final overall correction to what comes out from MIX. Very comon question is "should I invert channel in INPUT, MIX or OUTPUT?". 
As I'm working now to group features in more human friendly way I'd be happy to see OUTPUTS use-case (setup) other than hardware related mix-correction.

pfeerick, 2022-10-06:
https://github.com/EdgeTX/edgetx/pull/2085 and https://github.com/EdgeTX/edgetx/pull/2219 will have some bearing on this in 2.9, as once a name can be added to more things like will be done in #2085, the intention in #2219 is to take that further, and also show the name if one is configured - as is already being done there for GVs.

gagarinlg, 2022-10-06:
primary goal for me is here to have the channel numbers available for 2.8, so that we get less complaints about the UI.

PR #2464 (2022-10-09): fix(gui): add missing channel IDs to mixer and output tabs. 
Fixes #2073

Summary of changes: add missing channel ids to mixer and output tabs
This PR currently creates those ouputs:
[image]
[image]
[image]
[image]
commit 40ae2e22 (2022-10-09): fix(gui): add missing channel IDs to mixer and output tabs.  (#2464)

* fix(gui): add missing channel IDs to mixer and output tabs. Fixes #2073

* fix: input display and more space for names

* made output names fit on all display variants

* channe id display fine tuning

* Outputs with no name are bigger
... magnifying glass was not included with tx ;)
commit 457a967e (2022-10-10): fix(gui): add missing channel IDs to mixer and output tabs.  (#2464)

* fix(gui): add missing channel IDs to mixer and output tabs. Fixes #2073

* fix: input display and more space for names

* made output names fit on all display variants

* channe id display fine tuning

* Outputs with no name are bigger
... magnifying glass was not included with tx ;)
closed 2022-10-09 by pfeerick