EdgeTX/edgetx #2077: Bluetooth Master/Slave discovery not working in 2.8

dlktdr, 2022-06-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

On version 2.8, if you select Master/Bluetooth option,

The init button doesn't automatically switch to discover when the button is pressed and the module is detected.

If you back out of the screen and back in the button does show Discover. When you press discover it does send a request to do the discovery and the module does return the addresses, but no pop-up to be able to select them.

[image]

Probably just new UI issues

Also tested with the PR #2072 no change, cannot discover

### Expected Behavior

Should show discovered addresses

### Steps To Reproduce

Turn on radio, choose Master/Bluetooth, choose discover. 

Currently using the PARA module on a TX16s

### Version

Other (Please specify below)

### Transmitter

Radiomaster TX16S

raphaelcoeffic, 2022-06-22:
@dlktdr this is on the trainer screen, right?

dlktdr, 2022-06-22:
Yes it is. I was just looking at the commented code. Wish I had stayed up to date more with the LVGL changes to help out more, have some learning to do again.

Decided to make an issue here as it came up on RCG. Something others can find and follow.

raphaelcoeffic, 2022-06-22:
> Decided to make an issue here as it came up on RCG. Something others can find and follow.

Folks really need to realise there is no such thing as "release 2.8.0" as of now. It is a development version that is unfinished.

raphaelcoeffic, 2022-06-25:
fixed with c616204cb9e1e1b389

closed 2022-06-25 by raphaelcoeffic