EdgeTX/edgetx #2082: Channels Monitor does not show override after special function Play Track is modified

mh125, 2022-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After a special function with "Play Track" is modified or added, the Channels Monitor does not show any overrides anymore.

Tested with 2.7.0 and 2.7.1

### Expected Behavior

The Mixers Monitor always displays things correctly.

### Steps To Reproduce

1. Create Model
2. Create Special Function, SC down = Override CH1 with 100, on
3. Return to home, go to Channels Monitor, see CH1 is 100 when SC down. The override is working and shown.
4. Create Special Function, SA down (or any other) = Play Track 3d (or any other track)
5. Return to home, Go to Channels Monitor, see that the Channel Monitor does not show the override any more.

After a reboot, everything works fine again.

### Version

2.7.1

### Transmitter

Radiomaster Zorro

mh125, 2022-06-29:
This bug is not about "Overrides not shown", but rather "Channels Monitor" not available any more and "Mixers Monitor" shown instead. I'll modify the title.

Another way to reproduce:
1. Show Channels Monitor (with ENTER, one can switch between Channels Monitor and Mixers Monitor)
2. Go to settings and reduce the volume
3. Try to go back to Channels Monitor. It is impossible. Only Mixers Monitor is available.

philmoz, 2023-12-13:
Is this still an issue?