EdgeTX/edgetx #2083: Swedish special characters not displayed in Xlite with LVGL

ulfhedlund, 2022-06-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After the update to LVGL in main branch, the "special" Swedish characters (åäöÅÄÖ) are displayed incorrectly as blanks on Xlite b/w radio. For TX16S it works as expected. Possibly the same for other languages special characters as well but I have only tested with a build of the Swedish firmware for Xlite and run it in the simulator, so not tested on the radio yet.
I have attached a screenshot where "Ä" is missing in the top menu name MODELLINSTÄLLNING (model settings) and "ä" at the bottom Nedräkning (countdown).

### Expected Behavior

All characters should be displayed correctly.

### Steps To Reproduce

Compile main branch with "cmake -Wno-dev -DPCB=XLITE -DDEFAULT_MODE=2 -DGVARS=YES -DPPM_UNIT=US -DHELI=NO -DCMAKE_BUILD_TYPE=Release ../" and run in simulator.

### Version

Other (Please specify below)

### Transmitter

FrSky X-Lite / S / Pro

### Anything else?

[image]

ulfhedlund, 2022-06-22:
You also need an updated se.h file since some strings are missing in the main branch. I've not yet made a PR for this but have attached an updated file (se_h.txt).
[se_h.txt](https://github.com/EdgeTX/edgetx/files/8961061/se_h.txt)

raphaelcoeffic, 2022-06-23:
> You also need an updated se.h file since some strings are missing in the main branch. I've not yet made a PR for this but have attached an updated file (se_h.txt). [se_h.txt](https://github.com/EdgeTX/edgetx/files/8961061/se_h.txt)

Would be cool if you could actually make a PR for the translation file ;-)

ulfhedlund, 2022-06-23:
Absolutely, that's the plan but I need to work on the translations a bit more :)

ulfhedlund, 2022-06-25:
I have now created a PR (#2095), hope I did it right. I'm not at all comfortable with github.

ulfhedlund, 2022-07-04:
@rotorman has now merged PR #2095 to main :)

raphaelcoeffic, 2022-07-05:
@ulfhedlund so this issue here can be closed, right?

ulfhedlund, 2022-07-17:
Just tested with Danish translations on Xlite and it is the same there - special characters are not displayed correctly.

ulfhedlund, 2022-07-18:
Just tested with Danish translations on Xlite and it is the same there - special characters are not displayed correctly. For what it is worth...

ulfhedlund, 2022-08-30:
Is there anything I can do to help with this? I suppose this issue is the same for all B/W radios (I have tested Xlite, X7 and X9D+) and for other languages than Swedish too (Danish for example). If anything that does not include coding needs to be done perhaps I can help. I am certainly willing to try.

gagarinlg, 2022-08-30:
Maybe there is a bug in the utf-8 decoder

gagarinlg, 2022-08-30:
It look like the character lookup tables for some language are missing in
https://github.com/EdgeTX/edgetx/blob/main/radio/src/gui/common/stdlcd/utf8.cpp

ulfhedlund, 2022-09-01:
Thank you, I have a local environment to build software so I'll add those and test it out.

ulfhedlund, 2022-09-01:
@gagarinlg - I did as you suggested and this issue now seems to be fixed!
I have build firmware for X9D+, X7 and Xlite in Swedish and tested in simulator. All characters are now properly displayed.

@HThuren: I also added the Danish special characters and tested on X7 in simulator. Seems like the Danish characters also are OK now. :)

I'll add a PR for this as soon as possible.

HThuren, 2022-09-01:
@ulfhedlund Thank you, I wasn't aware of this issue. So great you has it fixed :-)

ulfhedlund, 2022-09-01:
@HThuren: Right now Danish characters are still displayed as blanks (" "). I tried fixing this but ended up removing the above lines, since the Danish special characters weren't displayed properly, but rather the Swedish equivalents. 

If you are happy with that, you can add the four lines above to utf8.cpp and make a PR. Svenska åäöÅÄÖ är kanske bättre än inget?

gagarinlg, 2022-09-01:
And the we look like idiots, because we can not add the correct Danish characters 🤔😜
We should fix the characters, can't be too hard.

HThuren, 2022-09-01:
Arh, now I understand. 
this section must be ok, but I'm not sure how it works, ie the order of charaters. Normally we use æøå, ÆØÅ, but ..
#elif defined(TRANSLATIONS_DA)
static wchar_t _utf8_lut[] = {
L'å', L'æ', L'ø', L'Å', L'Æ', L'Ø',
};

gagarinlg, 2022-09-01:
https://github.com/EdgeTX/edgetx/blob/main/tools/charset.py

There is no danish support in that file, so the danish special characters are not added.

HThuren, 2022-09-01:
OK, leave it for now, I add (and want to test) after this PR #2083 are added to main.

pfeerick, 2022-09-02:
@HThuren Basically what this means is for Danish, you'd need both the 

```
#elif defined(TRANSLATIONS_DA)
static wchar_t _utf8_lut[] = {
L'å', L'æ', L'ø', L'Å', L'Æ', L'Ø',
};
```

block you gave above in `radio/src/gui/common/stdlcd/utf8.cpp`

as well as a corresponding entry in 
https://github.com/EdgeTX/edgetx/blob/8aa2b55c503de70d5a7c0df1ca499597f3434613/tools/charset.py#L58-L72

for B&W radios to show those 'special' Danish characters properly.

closed 2022-09-02 by pfeerick

HThuren, 2022-09-07:
Hi @ulfhedlund 
Thank you for your support with this case. Danish characters are now drawn with Paint, and the files must exist in radio\src\fonts\sqt5 and radio\src\fonts\std and refered in fonts.cpp and utf8.cpp :-)

PR #2259 (2022-09-02): Added characters for TRANSLATIONS_SE
Fixes #2083

Summary of changes: 
Added characters for TRANSLATIONS_SE so that Swedish special characters are displayed on B/W radios.
Thank you @gagarinlg for identifying the cause of this issue.