EdgeTX/edgetx #2086: Qx7 access Baudrate setting missing

Skillexslv, 2022-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

The setting to change the baudrate is missing for the qx7 access

### Expected Behavior

there should be a setting to change the baud rate under the hardware tab in setting 

### Steps To Reproduce

the setting is just not there

### Version

2.7.1

### Transmitter

FrSky Q X7 / Q X7S (ACCESS)

pfeerick, 2022-06-23:
Do you mean for things like Crossfire, Ghost? As that is under the model -> external rf now as it's configured on a per model basis.

j-brubaker, 2022-06-26:
On my Horus X10S, I have the baud rate option visible when external module is CRSF (ELRS) but no baud rate option for GHST. Running 2.7.0. 

Could this be why I’ve been getting constant telemetry warnings flying Ghost recently? I don’t think I was getting them on 2.5 but can’t remember for sure. 

The X10S appears to require an inverter mod like the QX7.

pfeerick, 2022-06-26:
Ghost only supports one baud rate settting, so it was not actually valid that it be shown. This is a main part of the reason this setting was moved to model settings - as it is actually module specific as to 1) if a range is supported, 2) what that range is. i.e. GHOST the MPM is a fixed baud rate, as is Ghost. Crossfire is either 115k or 400k, but since ELRS also uses CRSF and supports up to 5M (depending on hardware) it needs to be as wide as the tx supports. As to why getting warnings now from GHOST, I'm not sure... it would probably depend specifically on the telemetry warnings given, and validation that there is no other hardware issues.

closed 2022-09-19 by pfeerick