EdgeTX/edgetx #2088: Global functions - cannot remove, just delete empty positions

Eldenroot, 2022-06-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

1) Create multiple items in the global functions menu
2) Long tap to open a menu and click on "delete" 
3) The selected global function is deleted but there is an empty space, you cannot remove i completely now (after deletion)
4) However if you add any new global function and use "remove" on long tap -> the empty space is deleted completely

### Expected Behavior

Add a new option "remove" for empty global functions

### Steps To Reproduce

See above

### Version

2.7.1

### Transmitter

Radiomaster TX16S

Eldenroot, 2022-06-23:
Also same for "Special functions" menu :)

Eldenroot, 2022-10-11:
Guys, can anybody reproduce this with 2.8? I think it is same...

raphaelcoeffic, 2022-10-12:
Somebody is trying is trying to sneak something into 2.8 🤭

pfeerick, 2022-10-12:
Or isn't actually using the RCs, since "delete" and "clear" are both present ;)

raphaelcoeffic, 2022-10-12:
This is how it works for me:
- `clear` basically makes the slot empty.
- `delete` make the slot empty and moves the entries bellow one up.

So I believe this works as designed (tested with current `main`). @Eldenroot am I missing something?

pfeerick, 2022-10-12:
Actually, the reason this wasn't commented further on is that you *can* delete (remove completely) and clear (just remove the content) both GVs and SFs in 2.7.1 also, so unless I'm missing something there's an #invalid tag coming for this issue soon :-P You can't delete LS, but that is completely different and would make life painful it that was allowed since it would basically screw up the LS numbering.

Eldenroot, 2022-10-12:
> Somebody is trying is trying to sneak something into 2.8 🤭

Nope, but this is my report and I wanted to be sure that it is still valid.

Eldenroot, 2022-10-12:
> Actually, the reason this wasn't commented further on is that you _can_ delete (remove completely) and clear (just remove the content) both GVs and SFs in 2.7.1 also (I just double and triple checked), so unless I'm missing something there's an #invalid tag coming for this issue soon :-P You can't delete LS, but that is completely different and would make life painful it that was allowed since it would basically screw up the LS numbering.
> 
> edit: Having said that, there was a delete refresh issue that was fixed a while back, but I can't pin down when exactly - I seem to think it was further back between 2.6 and 2.7- where delete would act like a clear, but if you left the screen and came back the screen would update properly. Or at least I think that is what it was.

Yeah, it has been a while, but your edit addition was that. But with 2.7/2.8 I was confused for clear and delete option. If you clear and the slot is empty (but isnstill here), you cannot after that delete this possition anymore. You have to again add some text amd then you get a new option to remove it. I think this is wrong.

raphaelcoeffic, 2022-10-12:
> you cannot after that delete this possition anymore. You have to again add some text and then you get a new option to remove it. I think this is wrong. You should be able to clear/delete anytime, am I wrong?

Let's call this "cosmetical". I believe this will go away when the screen is redone using the new LVGL flex layout (like inputs/mixers/etc). In that case you will see only the used slots anyway, and I believe order does not really give you anything here. So adding will be at the end of the list, no matter what, and there won't be a difference between "clear" and "delete" anymore.

Eldenroot, 2022-10-12:
Ok.

Do you have any list of unconverted pages and parts of gui into lvgl?

pfeerick, 2022-10-12:
Nope... just look at them... the ones that look clunky still are non-LVGL 😆 IIRC it's curves, functons (special/general), global variables, logical switches and telemetry. There have been minor LVGL touches to them, but only enough to make them work. Most other screens have all been moved over/refreshed. 

Ok, yes, I see what you mean, you can't delete after you've cleared, which is a difference issue entirely to the one before. And as Raphael said, should be resolved once this page is refreshed.

raphaelcoeffic, 2022-10-12:
> Do you have any list of unconverted pages and parts of gui into lvgl?

You can see it, as it's using the old layout:
- Global functions
- Curves (on-going work)
- Global variables
- Logical Switches
- Special Functions
- Custom scripts
- Telemetry

Eldenroot, 2022-10-12:
Thank you, I am glad that we have debugged my request :)

raphaelcoeffic, 2022-10-12:
LoL! Just created #2522 for the screens that need to be converted.

closed 2022-10-12 by raphaelcoeffic