EdgeTX/edgetx #2090: Curves - wasted space on the right side

Eldenroot, 2022-06-23:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

A lot of wasted and unused space on the right side.

### Describe the solution you'd like

We should display values, number of points etc.

### Describe alternatives you've considered

---

ulfhedlund, 2022-06-29:
The right hand side seems to be the only place to click to be able to edit and remove a curve. This took me a while to figure out. Maybe it is intentional but to me it is not intuitive.

Eldenroot, 2022-10-11:
#2508 - maybe these two could be connected

pfeerick, 2022-10-11:
Possibly. It's really up to whoever decides to work on these, as while it it related to curves, they are two completely different screens, and different levels of complexity to implement. You are more than welcome to contribute some code towards this.

gagarinlg, 2023-03-18:
Please check the current nightlies. The curves page hsa been redone.

closed 2023-03-18 by Eldenroot