EdgeTX/edgetx #2091: NV14 with ExpressLRS Status error

zandorsp, 2022-06-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When using the NV14 with external ExpressLRS module the status reports constantly errors

### Expected Behavior

The number of error don´t change

### Steps To Reproduce

1 - Connect an external ExpressLRS module
2 - Create/Select a model with CRFS external module
3 - See the status

### Version

2.7.1

### Transmitter

FlySky NV14

### Anything else?

[image]

pfeerick, 2022-07-03:
As in, it keeps incrementing when you switched to 115k, or that the number remains (you may also have noticed the hz dropped value dropped oddly also)? I would not go above 400k on the NV14 as I don't think the inverters on it are fast enough to do anything higher (relaibaly, that is). 921k seems mostly stable, but still has somewhat intermittent incrementing handset -> module error count so I wouldn't use it myself without a GPS failsafe recovery system. 

You may need to disconnect and reconnect the module after changing the baud rate so that the module re-detects the new baud rate as while the module *is* automatically power cycled, it seems that some ELRS modules have enough capacitance to stay partially powered during the reset cycle.

Neukom720, 2022-07-05:
NV14 edgetx 2.7.1 - yep i just battled with the same problem! Doesn't matter the baud rate, the errors just continuously go up with the baudrate just changing how quickly the errors increase, 115k-400k-921k-1.87k all work and I can access the elrs LUA script but the errors just start building up faster and faster as you increase the baud rate, at 1.87k the error is going up 100 every second!  3.75M and 5.25M don't even work and looses connection with the elrs module. Also the elrs Lua script reports 0/50 or zero out of whatever refresh rate your using which is strange because you think it would be dropping frames ya know. Just Flew on 400k using elrs Happymodel ES900tx@50hz and flew fine but it was just a long flight no fast moves or around anything. Both my Happymodel 2.4ghz elrs and my Full size Crossfire 2watt module has the same outcome. Any ideas? Its worry some...thanks guys as always for any help you can give us NV14 boyzzzz haha

Neukom720, 2022-07-05:
@pfeerick tried pulling the module out after changing baud rate, same outcome. As soon as you plug it back in the error count just continues right on going up....ill try to help troubleshoot if you guys have any other ideas

pfeerick, 2022-07-05:
> Also the elrs Lua script reports 0/50 or zero out of whatever refresh rate your using which is strange because you think it would be dropping frames ya know.

Unfortunately that's not how it works... these are the errors between the handset and the module - which is *usually* because the speed is pushed beyond what the radio hardware can operate at. It's not related to the control link between the module/transmitter and receiver. So you can have 100% module link/spot on hz rate (i.e. the Bad/Good indicator at the top shows `0/%SET-HZ-RATE%`, and errors on the handset side, and as long as it's a slow creep, because of the data rates we are talking about, they would never propagate through anyway. It's just something to be concerned about until the exact cause can be pinned down... i.e. if the inverters on that particular handset are slow, that's all good and fine... at least it won't get any worse... and you have the _possibility_ of replacing them. 

I'm using 2.4 ELRS, but do have some 900Mhz ELRS so will try that to see if there is any difference...

Neukom720, 2022-07-05:
@pfeerick tried all my modules, all do the errors, doesn't matter what, my happymodel elrs 2.4ghz and 900mhz, my Ghost Lite flashed to elrs using jr bay adapter, my full size 2watt crossfire, my r9m lite pro on elrs using a jr bay adapter, all do the error thing at whatever baud rate you choose, just trying to save you time on trying different modules cause its def radio side. Also tried elrs v3.0.0rc1 and 2.5.8 both same outcome. Haven't used my tx16s in a long time cause I only use it for fixed wing stuff but I'll dust it off and flash edgetx to verify as well.

raphaelcoeffic, 2022-07-05:
The error counter you see is actually accurate, and most of these are so called "Framing Errors":
[image]

I believe this happens because there is no pull-down resistor neither on the external radio side nor on the external module side. Please note that this "inverted UART", so that IDLE is LOW.

pfeerick, 2022-07-06:
Oh, interesting... yeah, that decay on the end would certainly do it, and as you say, is harmless. I'll have to see if that's what I'm getting. Interesting that it doesn't seem to be happening at the lower baud rates as much in mine... maybe I'm not waiting long enough, or this module actually does have a pulldown or some other filtering.

zandorsp, 2022-07-06:
> The error counter you see is actually accurate, and most of these are so called "Framing Errors": [image]
> 
> I believe this happens because there is no pull-down resistor neither on the external radio side nor on the external module side. Please note that this is "inverted UART", so that IDLE is LOW.
> 
> That being said, it should be pretty much harmless here, as the errors are only artefacts and do not disturb real payload.

That's means that I need a pull-down resistor on S.PORT line in the radio or in the module?

raphaelcoeffic, 2022-07-06:
> That means that I need a pull-down resistor on S.PORT line in the radio or in the module?

Actually it would be better on the MCU side (between MCU and configurable inverter). We're checking the NV14 hardware, as there seems to be some not-connected pads that were meant for a pull-up resistor at the right place.

zandorsp, 2022-09-14:
I don't know what you made in the nightly releases, but it's seen to work now without errors increasing in status.

pfeerick, 2022-09-14:
Oh sweet! :) Most likely some changes for the EL18 also effect the NV14 (but in a good way). Thanks for the update! 🥳

closed 2022-09-14 by pfeerick

guimeraalex020, 2022-12-16:
Hey, i have the same problem with the nv14, and the updates aren't solving the problem. What version of edgetx and elrs do you had to solve the problem? Thanks
@zandorsp

macfly1fr, 2022-12-17:
> Hey, i have the same problem with the nv14, and the updates aren't solving the problem. What version of edgetx and elrs do you had to solve the problem? Thanks @zandorsp

I confirm i have the same issue with all edgetx version. It was not a "real problem" until now when i updated my Tx HappyModel ES24tx 2400 with the latest elrs firmware version 3.1.1 -> when i run elrs's lua script V3 it freeze (in fact it is very very sloowww so unusable ) .
It worked fine (always status error) with all previous elrs firmware (ok with elrs 3.0.1 too ).So i had to flash back to elrs 3.0.1 firmware version... I suppose this issue is really due to status error continuously of the Nirvana with external module in edgetx.

guimeraalex020, 2022-12-17:
In my case the lua script is working correctly, i don't understand how the lua is communicating correctly but the radio can't without errors

macfly1fr, 2022-12-17:
> 

> In my case the lua script is working correctly, i don't understand how the lua is communicating correctly but the radio can't without errors

Your lua script works correctly with elrs "3.1.1" ?

zandorsp, 2022-12-18:
> Hey, i have the same problem with the nv14, and the updates aren't solving the problem. What version of edgetx and elrs do you had to solve the problem? Thanks @zandorsp

I'm now using EdgeTX 2.8.0 with elrs  3.1.1. The status show some errors, but really not like my first post. 
[image]

gagarinlg, 2022-12-18:
A few errors are normal, they arise when everything is booting up.

guimeraalex020, 2022-12-18:
I don't have a few, i have a increasing error number

gagarinlg, 2022-12-18:
@guimeraalex020 the few errors zandorsp has are OK.

A slightly increasing error number does not stop the Lua script from working.
Most people seem to get better results with EdgeTX 2.8.0. Which version do you have on your radio?

guimeraalex020, 2022-12-18:
I had 2.7.0 but already upgraded to 2.8.0 and still the same, up to 20.000 errors and increasing.

rotorman, 2022-12-18:
Do the errors disappear or at least lower, when you use lower baudrate? E.g. 400 kBaud?

guimeraalex020, 2022-12-18:
They speed down, but they never stop

macfly1fr, 2022-12-22:
I have the same... errors never stop
My only real issue i that in my case i can't use elrs lua script v3 with module having elrs firmware > 3.0.1

zandorsp, 2022-12-23:
> I have the same... errors never stop My only real issue i that in my case i can't use elrs lua script v3 with module having elrs firmware > 3.0.1

My NV14 radio was the pull-up resistor from s.port line removed. I don't know really if this cause your problems.
[image]

Reference: https://docs.google.com/document/d/1SRegkGT0wnFcTbT9YFNwIuluz5aupOivjGYq4cDiHjE/edit

Neukom720, 2022-12-23:
> They speed down, but they never stop

Yep I still have this problem, acts the same way, accessing the ELRS lua script also gets screwed up in the higher baudrates or doesn't work at all

Neukom720, 2022-12-23:
Has anyone verified that removing the pull up resistor from the s.Port line fixes this issue? I'm totally down to remove it as long as it has no I'll effects and removes this issue

mmaaddzz, 2022-12-24:
I can confirm, after removing that resistor lua script working perfect and errors count is about zero all time.

Neukom720, 2022-12-24:
@mmaaddzz 
Thanks man, I'll do that tomorrow and hopefully report back with my success as well!!

guimeraalex020, 2022-12-24:
I did it and now its working!!!!! it was only removing the  pull-up resistor

rotorman, 2022-12-25:
The resistor in question would be the same as R14 in publicly available EL18 schem: https://raw.githubusercontent.com/open-flysky/EL18-_OpenSourceMaterial/main/sch/EL18-V1.0%E5%8E%9F%E7%90%86%E5%9B%BE(sch)%E5%8F%8A%E7%82%B9%E4%BD%8D%E5%9B%BE.pdf (on EL18 this resistor is not populated in factory).

In case on NV14, the S.PORT line is similarly routed, looking at the schematic, I wonder why it helps to desolder it. Nevertheless great to hear that this helps to solve it.
@richardclli do you know if NV14 S.PORT schem is different to EL18 or have an idea why removing this pull-up makes a difference?

macfly1fr, 2022-12-26:
> > I have the same... errors never stop My only real issue i that in my case i can't use elrs lua script v3 with module having elrs firmware > 3.0.1
> 
> My NV14 radio was the pull-up resistor from s.port line removed. I don't know really if that's what causes your problems. [image]
> 
> Reference: https://docs.google.com/document/d/1SRegkGT0wnFcTbT9YFNwIuluz5aupOivjGYq4cDiHjE/edit

Thank you for the information.
I removed this resistor and.... I confirm all problems are solved.
Error status is now very stable , i don't have any issues with last elrs firmware and lua script.

Thank you everybody for the help.

OG-ALviK, 2023-01-06:
So, for the NV14 users who are having these tipes of issues, to fix them just remove the resistor?

zandorsp, 2023-01-06:
> So, for the NV14 users who are having these tipes of issues, to fix them just remove the resistor?

Yes, that's it.

DeadSeriious, 2025-04-24:
**Question**

I am trying to "revive" my Nirvana NV-14 here.... I never really ended up using it at all because I pretty shortly after got a TX16S and it's been my daily driver ever since.

My question though, is I tried updating EdgeTX to ditch that awful UFPV fork and get a normal, up-to-date firmware version running, that seems to be all fine and dandy now. However, using my ELRS module has been loaded with problems. The biggest of which being if I'm running 250mw output power, the radio "dies" almost instantly. I read that this thing cannot support more than 100mw output without having an external power supply/battery. Is this accurate today? Or has anyone managed to find a work around to getting it working without relying on an external pack at >100mw...? 

The other issue that I've been seeing pop up here and there while researching for fixes is issues with ELRS and the gimbal polling rate? What exactly is the issue there and is there anything I can do about THAT issue?

And finally..... does anyone have ANY experience in general with using this radio in 2024/2025 using ELRS? If so, I'd like to know what your experience thus far has been? What (many) hurdles you've had to jump through, etc.

Thanks a bunch. <3 It's too bad this radio was a complete shitstorm during it's time. It had so much potential to be awesome and it was just fumbled it seems, by everyone involved in bringing it to fruition. x_x I hope that the community has had some luck in DIY'ing some fixes to this thing over the years since it's release....

richardclli, 2025-04-24:
> **Question**
>
> I am trying to "revive" my Nirvana NV-14 here.... I never really ended up using it at all because I pretty shortly after got a TX16S and it's been my daily driver ever since.
>
> My [...]

NV14 internal power output is not stable enough to drive external ELRS module.  This can cause a blackout in NV14 if the battery voltage is low + ELRS power consumption is high.