EdgeTX/edgetx #2092: Support GVars in calculated telemetry sources /OTX/

Eldenroot, 2022-06-24:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This seems to be a good idea to add an ability to use Gvars in telemetry.

### Describe the solution you'd like

Add a support to use Gvars in Telemetry calculations

https://github.com/opentx/opentx/pull/8937

functionpointer, 2022-06-25:
I am actually working on porting this, as there doesn't seem to be any reaction to this from the OpenTX devs.

Unfortunately, calculated sensors don't seem to work at all in simulator builds currently. They work in `2.7.1`, but don't on `main`. Any calculated sensors just have value `0`.

I have bisected and got the following result (i had to skip a lot due to `fifo.h` missing)
```
There are only 'skip'ped commits left to test.
The first bad commit could be any of:
1017172350e65560ec27a97634bf883b8fa37506
885c0d1a2e8653e6ed08c109a3519c2dc6e85200
d318f6d9bd52fe98ab9f63db4ab131f7a6b504d6
dcc8902f175180291c50a2a4f37bea3cd4d24157
ead920bf1d5cc488227e80898f5a5065c901137a
ee10cf9ff854572902c8519acd5818dd1f30c918
0613a291224ebc6e942008fa6c23cc9232101980
dbb9a5e4b2b710c3ca5f4fee9e987753c63d2c00
0927847adb7fe79672218b592b94891373d3e29a
8a988b7f882fa97fab8527096d7bf9114a282a8d
a994fc708a0692c64d8a0d74fbdbdf790025173b
c9594390e278c265e2e8507f794120d1637bbf17
59191f37c3b7bc52b48356f38b24f912e388ba73
5e6cf2e79cfdf325b7260beeccf775aee53af0e9
221d7177e49717943f97a21cd056ebf523a9bcbc
```

raphaelcoeffic, 2022-06-25:
@functionpointer The issue is that we use now FreeRTOS Software timers to trigger telemetry calculations, which are unfortunately not supported (yet?) in the simulator.

closed 2026-11-09 by pfeerick