EdgeTX/edgetx #2094: Frsky 10s with IRSM board

alant101, 2022-06-25:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No available at present

### Describe the solution you'd like

There seems to be various builds of Edgetx noheli available, but none for the Horus 10 with the replacements IRSM board for ACCESS. This would be appreciated if possible

### Describe alternatives you've considered

None

### Additional context

None

raphaelcoeffic, 2022-06-25:
Please use the discussion related to ask for specific builds ;-)

closed 2022-06-25 by raphaelcoeffic

pfeerick, 2022-06-26:
Use the regular X10 build, and use Radio Settings -> Hardware to set the Internal RF type to IRSM.