EdgeTX/edgetx #2102: Horus 10 with ISRM RF mode

alant101, 2022-06-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

NO option to select ACCESS in menu Horus10

### Describe the solution you'd like

I was recently told that this was an option to select ACCESS in the menu and after installing 2.7.1 found that this is not to be correct. D8,D16 and L12 only, if possible could this be added soon. Thank you

### Describe alternatives you've considered

None

alant101, 2022-06-27:
To save any confusion this a Horus 10 which has now a had IRSM fitted as an aftermarket upgrade

pfeerick, 2022-07-01:
Maybe this video will help you see the difference and *what* setting you should be altering to get internal ISRM and ACCESS option ;) You can see by default it is set to XJT, and it was changed to ISRM and ACCESS became an option. 

https://user-images.githubusercontent.com/5500713/176802233-3dddb45f-c885-4aa9-acde-d10a2f3eb0ab.mp4

pfeerick, 2022-08-12:
Closing as no follow-up and method of changing internal RF was shown.

closed 2022-08-12 by pfeerick