EdgeTX/edgetx #2105: Prompt when disabling extended trim if trim value is outside normal range

RJMcInty, 2022-06-28:
from @pfeerick:

This fixes what could be perceived as a bug... whereby if you change from extended trim, to 'normal' trim range,  and the trim was in the extended range, it would wind back until it's in range (but not let you go back out of range). Now, it jumps back to the maximum for the trim on the first click. IMO this really needs to be caught when the option is changed, and be the focus of another PR. i.e. when you untick "extended trim" and you are in the extended range, a caution should come up, only allowing you to reset the trim or cancel, forcing you to move it back first if it needs to be done gradually.

_Originally posted by @pfeerick in https://github.com/EdgeTX/edgetx/pull/1955#pullrequestreview-1019948795_

RJMcInty, 2022-06-28:
I'm interested in looking at this issue, it seems straightforward enough, except I'm struggling a bit with wrapping my head around the UI code. Can someone point me to code that's executed when this option is changed so I can look around?

pfeerick, 2022-06-28:
I'm think relevant sections are 

for colorlcd:
 https://github.com/EdgeTX/edgetx/blob/ce8cc0b540dc287c41d52e1c601b37e762e8c35c/radio/src/gui/colorlcd/trims_setup.cpp#L66-L69

for 128x64 B&W
https://github.com/EdgeTX/edgetx/blob/ce8cc0b540dc287c41d52e1c601b37e762e8c35c/radio/src/gui/128x64/model_setup.cpp#L697-L711

and for 212x64 B&W
https://github.com/EdgeTX/edgetx/blob/ce8cc0b540dc287c41d52e1c601b37e762e8c35c/radio/src/gui/212x64/model_setup.cpp#L601-L615

RJMcInty, 2022-06-28:
Thanks! I'll take a look at this.

raphaelcoeffic, 2022-06-28:
> I'm interested in looking at this issue, it seems straightforward enough, except I'm struggling a bit with wrapping my head around the UI code. Can someone point me to code that's executed when this option is changed so I can look around?

On the colorlcd UI, you can just “register” a event callback with VALUE_CHANGED on the component itself. Sometimes there is a handler which can be set with libopenui as well.